openwrt/packages #10829: ModemManager : netifd not waiting for ModemManager to be connected to dbus

jonesmz, 2019-12-18:
Maintainer: @nickberry17

Description:

I'm running into problems where netifd is attempting to bring up the modem before ModemManager is actually running.

This results in the call to

```
modemstatus=$(mmcli --modem="${device}" --output-keyvalue)
```

failing.

Then when modemmanager later comes online, the hotplug scripts trigger, and cause modemmanager to lookup the interface associated with the modem that it find's /sys/ path, and then signals netifd to bring the interface back online.

Is there no mechanism in netifd to just... wait?

feckert, 2019-12-19:
We could wait in the netifd modemmanger protocolhandler instead of failing.
https://github.com/openwrt/packages/blob/master/net/modemmanager/files/modemmanager.proto#L332-L342
Have a look at https://github.com/openwrt/openwrt/blob/master/package/network/utils/uqmi/files/lib/netifd/proto/qmi.sh#L82-L92
If the sim is not initialized  after some timeout we could do a restart.

nickberry17, 2019-12-19:
> We could wait in the netifd modemmanger protocolhandler instead of failing.
> https://github.com/openwrt/packages/blob/master/net/modemmanager/files/modemmanager.proto#L332-L342
> Have a look at [...]

Nice idea, I grabbed that code and put it into the MM proto handler.  Please check it here and test if you can.  I wonder if it's sufficient to restart the interface (as it currently does) or if we need to restart the MM service.  Also, question about the errors, like `MODEM_NOT_INITIALIZED`, do we need to define them somewhere?
https://github.com/openwrt/packages/pull/10837/files

feckert, 2019-12-20:
The value `MODEM_NIT_INITIALIZED` is shown in the ubus status call of this interface. 
The value can be specified more precisely in the LuCI, what this means exactly.
For example https://github.com/openwrt/luci/blob/master/protocols/luci-proto-ncm/htdocs/luci-static/resources/protocol/ncm.js#L21-L27

aleksander0m, 2020-01-14:
If netifd starts and MM isn't around yet, the protocol handler will end up with the mmcli calls failing as the reporter said, and the protocol handler will end up returning DEVICE_NOT_MANAGED. Then, once MM starts, the interface would be made available and netifd will attempt to bring the interface up, and the protocol handler will succeed.

I'm not truly sure what the root issue is really; is there any case where due to this mismatch the modem ends up not handled or something? Or is this just so that neitfd protocol handler explicitly waits for MM to be around? If we don't want to have netifd wait for MM to be in the bus (don't think we would want that), I would not do any additional explicit wait either.

The initial setup done by netifd without any timeout is totally fine because it needs to have the initial state of the interface (`proto_set_available "${interface}" 0`).

The case with the qmi protocol handler is very different, because the wait there is to wait for the SIM to be initialized, and if the timeout happens the modem is considered unusable. In the case of MM, that logic is handled by MM itself. In other words; if you remove the timeout in the qmi protocol handler, you're losing functionality; but if we don't have a explicit timeout in the MM protocol handler I don't think we lose anything at all.

Not sure if I'm missing something really. I'd love to know if there is any real bug being fixed here.

jonesmz, 2020-01-30:
At least on my setup, this is not the case.

Netifd tries to start the modem, fails out because modemmanager isn't ready (for whatever reason) and then does not try again unless something else calls "ifup wwan".

I'm using OpenWRT 18.06.1, however, so maybe this is netifd behavior that has been changed since?

I've worked around this problem by having another script running in a while loop that checks such things as:

- the existence of a modem via the sysfs path of a modem being found
- `mmcli -m $sysfspath` giving me a non-error response
- `mmcli -m $sysfspath` showing the "state" of the modem as not "connected"
- `ubus call network.interface.wwan status | jsonfilter -e "@.up" = "false"`

and if those hold true, then i call `ifup wwan` which reminds netifd that it should do it's job.

aleksander0m, 2020-05-15:
> Netifd tries to start the modem, fails out because modemmanager isn't ready (for whatever reason) and then does not try again unless something else calls "ifup wwan".

Netifd shouldn't try to start the connection until the interface is reported as available, and that should only happen once the modem has been exposed by MM. If the interface is reported as available, and the modem is exposed by MM in DBus, and the start operation still fails, is it really because MM isn't ready yet? That is what we need to see.

One thing to note though: a ifup failure that is caused by e.g. a connection failure to the mobile network isn't automatically retried by netifd (I believe); I think this is what could be causing the root problem.

jonesmz, 2020-05-15:
When I say that modemmanager isn't ready, i'm not saying that the modemmanager service isn't ready. I'm saying "Modemmanager, the modem, and so on" is not ready, for whatever reason.

At least on my system, netifd isn't trying to start things up after the first failure. So I have to have a script that actively pokes things to make sure I get a connection eventually.

If you have specific logs that you want me to capture, i will.

aleksander0m, 2020-05-15:
Yep, I understand what you mean now, and you're right. A failure to connect the modem in netifd would not be retried unless explicitly requested by the user. In the same way that a network-initiated disconnect of the modem may not be detected by netifd itself. This is a big open point in the protocol I'm afraid, and not totally sure how to properly solve it in netifd and/or ModemManager.

feckert, 2023-09-13:
@aleksander0m @mips171 
I felled over this pullrequtest from @dangowrt  while cleaning up, which should fix the problem at least for an existing connection?

https://github.com/openwrt/packages/pull/20760/commits/c646b6f34b43619811597f635cc657dbb26ecf33

feckert, 2024-06-12:
Should be fixed with https://github.com/openwrt/packages/pull/24370

closed 2024-06-12 by feckert