openwrt/packages #10841: jupp not compiling on mips16 targets

neheb, 2019-12-19:
Maintainer: @mirabilos 
Environment: master

Example: https://downloads.openwrt.org/snapshots/faillogs/mips_24kc/packages/jupp/compile.txt

This happened with the switch to GCC8.

My theory as to why this is happening is this: https://github.com/mirabilos/jupp/blob/coverity_scan/config.h#L124

The generated assembly is MIPS32 while it should be MIPS16, hence the error.

Maybe it needs minor tweaking. I don't know enough about the code to judge it.

mirabilos, 2019-12-19:
No, that’s definitely not the cause (it’s not in the first .c file that it fails, *and* the error message does not match).

`Error: branch to a symbol in another ISA mode`

Guess: one of the libc headers contains inline assembly, and something jumps there.

Can you run the following two commands…

    mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I. -I. -DJOERC="\"/etc/jupp/\""  -I/builder/shared-workdir/build/sdk/staging_dir/target-mips_24kc_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include/fortify -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include  -D_GNU_SOURCE  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-mips_24kc_musl/jupp=jupp -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT bw.o -MD -MP -MF ".deps/bw.Tpo" -E -dD -o bw.i bw.c

and

    mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I. -I. -DJOERC="\"/etc/jupp/\""  -I/builder/shared-workdir/build/sdk/staging_dir/target-mips_24kc_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include/fortify -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include  -D_GNU_SOURCE  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-mips_24kc_musl/jupp=jupp -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT bw.o -MD -MP -MF ".deps/bw.Tpo" -S -o bw.s bw.c

… and attach the two files `bw.i` (preprocessed) and `bw.s` (compiled to assembly)?

mirabilos, 2019-12-19:
You could also test your hypothesis:

    mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I. -I. -DJOERC="\"/etc/jupp/\""  -I/builder/shared-workdir/build/sdk/staging_dir/target-mips_24kc_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include/fortify -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include  -D_GNU_SOURCE  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-mips_24kc_musl/jupp=jupp -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT bw.o -MD -MP -MF ".deps/bw.Tpo" -c -o bw.o bw.c

… in the build chroot should fail with the same error, and…

    mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I. -I. -DJOERC="\"/etc/jupp/\""  -I/builder/shared-workdir/build/sdk/staging_dir/target-mips_24kc_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include/fortify -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/include  -D_GNU_SOURCE  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-mips_24kc_musl/jupp=jupp -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT bw.o -MD -MP -MF ".deps/bw.Tpo" -c -o bw.o bw.c -DMKSH_DONT_EMIT_IDSTRING

… should succeed.

mirabilos, 2019-12-19:
Also check if GCC has the patch from https://gcc.gnu.org/ml/gcc-patches/2016-11/msg01061.html applied… if your GCC is old (8 is old, I think?) and binutils new (2.28) they might not like each other.

neheb, 2019-12-19:
Locally, I'm using GCC9 and Binutils 2.32. I don't think it's that.

I tested that hypothesis: The former failed and the latter succeeded.

Still need the files?

mirabilos, 2019-12-19:
The assembly takes care to switch to a separate ELF section for the string and then switch back, so it’s a no-op for the code flow.

I think it’s very much that GCC issue.

            beqz    $2,$L127
    $L53:
    #APP
     # 10 "bw-lgen.inc" 1
            .section .comment
            .ascii  "@(""#)rcsid_bw_lgen_inc: "
            .asciz  "$MirOS: contrib/code/jupp/bw-lgen.inc,v 1.5 2018/06/28 03:11:18 tg Exp $"
            .previous
     # 0 "" 2
    #NO_APP
            lw      $2,56($sp)

See how `$L53` is not followed by `.insn` nor any opcode? This marks `$L53` as not of the MIPS16 ISA.

Downgrading binutils would cause the thing to work, as this is basically a false-positive of a binutils check caused by a missing thing in GCC.

neheb, 2019-12-19:
Most likely a GCC bug. binutils did not get bumped to 2.32 on the buildbots. Even if bumped, it still fails locally.

What's the difference between the asm version and the LLVM one?

mirabilos, 2019-12-19:
The ASM one ends in the ELF section `.comment`, which can be stripped if space is tight, but would be present in a debugging or regular non-embedded build, which helps identify the exact code.

The version for compilers without good inline assembly support (LLVM+Clang used to not have it, might even be they do now) places it as constant string in `.rodata` together with all other strings, and are not strippable. (We’re talking about a few hundred bytes here, but…)

mirabilos, 2020-01-04:
I’ve added a workaround to jupp (which will be part of the next release) to avoid the `asm` when the code is built with GCC targetting MIPS16. (We can later amend that to include a GCC version check, when we have known fixed and broken versions.)

http://www.mirbsd.org/cvs.cgi/contrib/code/jupp/config.h.diff?r1=1.14;r2=1.15

You may wish to include it as a local patch if urgent; however, I intend to do a release to ship all the accumulated changes, probably some time this month.

neheb, 2020-01-09:
It's fine. It seems it's a GCC8+ bug. 19.07 and below don't have this problem.

mirabilos, 2020-01-13:
I wonder whether GCC devs will fix it. The workaround will ship with the next jupp release anyway, I adjusted the version to exclude GCC ≤ 7.

cotequeiroz, 2020-05-13:
Fixed by #12154

closed 2020-05-13 by cotequeiroz