openwrt/packages #10851: samba4: can i choose the stable samba version when compiling openwrt?

siwind, 2019-12-21:
Maintainer: @Andy2244
Compile tested:  MIPS/mipsel_24kc   openwrt HEAD 
Run tested:  Newifi D2 (router)

Description:

I often compile the master branch of openwrt, and also need the samba4 package to work with.
But every time the source code is updated (feed update), the samba 4 version will change, and Samba4 often leads to high CPU utilization of my router  , or system crash.

So is there any way to stabilise  the version of samba4?
for example
The latest stable version of samba4 4.9 series is: 4.9.17
The latest stable version of samba4 4.10 series is: 4.10.11

```
Formating code blocks by wrapping them with pairs of ```
```

Andy2244, 2019-12-21:
Latest stable is 4.11.4, which is in master now, mainly because 4.11 is the new python3 only version. In general 4.10 had a lot of issues and while 4.9 is tested longer 4.9/10/11 all have the same smb3 fileserver core code, which did not really change much in the last years.

I did add a new priority option via proc, so try lower smbd priority a bit might help if you really crash because of samba.
Config option:
```
config procd 'extra'
	option samba_nice '3'
```

PS: I also try to stabilize the samba alternative `cifsd server` so maybe try this one, since its more suited for low end devices.

Andy2244, 2020-01-07:
I just added `smbd` _(ex cifsd)_ to 19.07, so try this for a low end device if samba crashes, other than that we still have samba3 in base.

If this answers your question please consider closing the issue.

neheb, 2020-01-14:
The answer is no.

closed 2020-01-14 by neheb

siwind, 2020-03-18:
thanks very much~