openwrt/packages #10879: strongswan: missing deprecated ipsec.conf

swg0101, 2019-12-27:
Maintainer: @stintel 
Environment: master

Description: missing deprecated ipsec.conf

ipsec.conf has been deprecated for some time and is no longer maintained. The configuration for the newer versions are maintained via the vici plugin and by swanctl. However, it looks like disabling the stroke and ipsec plugins and enabling vici and swanctl will lead to the ipsec.conf not being copied (which is expected since it is no longer used). The install process crashes since ipsec.conf is still referenced as a configuration file to be installed.

```time: package/feeds/packages/xtables-addons/compile#0.52#0.21#1.00
rm -rf /root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/.pkgdir/strongswan.installed /root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/.pkgdir/strongswan
mkdir -p /root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/.pkgdir/strongswan
install -d -m0755 /root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/.pkgdir/strongswan/etc
install -m0600 /root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/ipkg-install/etc/ipsec.conf /root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/.pkgdir/strongswan/etc/
install: cannot stat '/root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/ipkg-install/etc/ipsec.conf': No such file or directory
Makefile:590: recipe for target '/root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/.pkgdir/strongswan.installed' failed
make[3]: *** [/root/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/strongswan-5.8.1/.pkgdir/strongswan.installed] Error 1
make[3]: Leaving directory '/root/openwrt/feeds/packages/net/strongswan'
time: package/feeds/packages/strongswan/compile#0.40#0.13#0.50
package/Makefile:111: recipe for target 'package/feeds/packages/strongswan/compile' failed
make[2]: *** [package/feeds/packages/strongswan/compile] Error 2
```

stintel, 2019-12-30:
Please try fe00850c888f65f4c0bf4db4a27b9d20bfeaa782. This is a very quick attempt to fix your build problem. A proper fix will have to wait until after the holidays.

I looked into adding an init script to load the connections via swanctl some time ago, but did not get very far. Here is what I currently have:
```
#!/bin/sh /etc/rc.common

START=99

USE_PROCD=1

load_conns() {
        i=0
        while [ $i -lt 10 ]; do
                res=$(/usr/sbin/swanctl --load-conns 2>/dev/null)
                [ $? -eq 0 ] && {
                        logger -t swanctl "$res"
                        return 0
                }
                let i=i+1
        done
}

start_service() {
        load_conns
}

reload_service() {
        load_conns
}```

It's quite ugly, but I don't see a better way right now. Feel free to suggest improvements :-)

swg0101, 2019-12-30:
I think the issue I am having is that the Makefile will not let the build complete because ipsec.conf is currently referenced by the main package and it is not copied unless the stroke and ipsec plugins are turned on. I will probably see if I have some time to take a look at it further...

swg0101, 2019-12-31:
I just realized that the change was pointing to a different commit - I will play around with it and the init script to see if I can get the connections to load properly. Thanks.

swg0101, 2020-01-13:
@stintel One thing I noticed without starter/stroke/ipsec was that charon needs to be started manually, otherwise any swanctl commands will fail (and swanctl itself does not start the charon daemon itself). This most likely means that there will need to be a different init script to start charon manually, and then perhaps a different init script to manage the actual loading of the connections.

Perhaps it is possible to do a combo script that starts and manage charon in the background, although doing it this way breaks the procd tracking and adds complexity.

Out of curiosity, did you plan to integrate the new swanctl config format with UCI or are you just going to keep it pointed at swanctl.conf? Given that the format is JSON like but not fully JSON, generating something like that may be a bit complex.

Thoughts?

mewsf, 2020-01-16:
For the startup script, I think we can refer to [Systemd Script](https://github.com/strongswan/strongswan/blob/master/init/systemd/strongswan.service.in)  from Strongswan official repo, it starts charon daemon first and then load config using swanctl.

And for me, UCI integration isn't that important, I edit config files manually.

In the strongswan package, strongswan-isakmp directly depends ipsec and stroke, maybe adding extra packages like strongswan-starter to distinguish from strongswan-swanctl is better.

Also, strongswan 5.8.2 is out, any chance we get this to openwrt soon? Thanks.

swg0101, 2020-01-16:
OpenWrt uses procd style init scripts, which are drastically different than Systemd style scripts and cannot be reused (unfortunately).

One thing I noticed about modifying config files directly (like I have been doing) is that the config files can get emptied on an unsafe shutdown/unplug/power outage. This can cause some issues if you are reliant on a remote VPN link and suddenly the site loses power and the VPN config is wiped and you can no longer get back in when the power is restored.

Problem with ipsec.conf is that it is deprecated and certain options are no longer available via that file... Bummer!

stintel, 2020-01-16:
> Also, strongswan 5.8.2 is out, any chance we get this to openwrt soon? Thanks.

f68f23f094e5b6efb9652b3d2a81916035d285b0

stintel, 2020-01-16:
> Out of curiosity, did you plan to integrate the new swanctl config format with UCI or are you just going to keep it pointed at swanctl.conf? Given that the format is JSON like but not fully JSON, generating something like that may be a bit complex.

I have no plans to do so right now. The UCI support was contributed by someone else, and I am not using it.

stintel, 2020-01-16:
And for swanctl support without UCI integration I will try to come up with something. First day I'm back home from holidays and without so present, started catching up with things!

swg0101, 2020-01-17:
> I have no plans to do so right now. The UCI support was contributed by someone else, and I am not using it.

Interesting, did you have troubles with your swanctl.conf being wiped clean on unsafe shutdowns? I know it certainly happens on stuff like strongswan.conf and ipsec.conf/ipsec.secrets...

stintel, 2020-01-17:
> Interesting, did you have troubles with your swanctl.conf being wiped clean on unsafe shutdowns? I know it certainly happens on stuff like strongswan.conf and ipsec.conf/ipsec.secrets...

I don't recall any of my strongswan files ever being wiped since I started using it on OpenWrt back in early 2014.

swg0101, 2020-01-17:
> I don't recall any of my strongswan files ever being wiped since I started using it on OpenWrt back in early 2014.

Although I don't suppose that you unplug the power from your router very often do you? I have a travel router in which this happens to all the time. Perhaps this is the thing when configs are stored on UBIFS.

mewsf, 2020-01-20:
> 
> 
> Please try [fe00850](https://github.com/openwrt/packages/commit/fe00850c888f65f4c0bf4db4a27b9d20bfeaa782). This is a very quick attempt to fix your build problem. A proper fix will have to wait until after the holidays.
> 
> I looked into adding an init script to load the connections via swanctl some time ago, but did not get very far. Here is what I currently have:
> 
> ```
> #!/bin/sh /etc/rc.common
> 
> START=99
> 
> USE_PROCD=1
> 
> load_conns() {
>         i=0
>         while [ $i -lt 10 ]; do
>                 res=$(/usr/sbin/swanctl --load-conns 2>/dev/null)
>                 [ $? -eq 0 ] && {
>                         logger -t swanctl "$res"
>                         return 0
>                 }
>                 let i=i+1
>         done
> }
> 
> start_service() {
>         load_conns
> }
> 
> reload_service() {
>         load_conns
> }
> ```
> 
> It's quite ugly, but I don't see a better way right now. Feel free to suggest improvements...
> 
> For now, if you need something that works, I would recommend to stick with ipsec.conf.

I'm not familiar with openwrt's procd system, but as I've known, it have to run `/usr/lib/ipsec/charon` in foreground first, then execute `swanctl --load-all --noprompt`. When reloading the service it should run `swanctl --reload` then `swanctl --load-all --noprompt`. I don't know if procd has something likes `ExecStartPost` like systemd, or maybe splitting charon and swanctl into two init scripts is also an idea.

swg0101, 2020-01-26:
In the meantime, I am just putting the following in the Startup script, seems to work fine:

/usr/lib/ipsec/charon &
sleep 2
swanctl -q

mewsf, 2020-02-19:
I saw a init script and I think it can be used with swanctl+vici:
```
#!/bin/sh /etc/rc.common
# Copyright (C) 2013 OpenWrt.org

USE_PROCD=1

START=90

EXEC=/usr/lib/ipsec/charon

start_service() {
  procd_open_instance
  procd_set_param command $EXEC
  procd_set_param stdout 1 
  procd_set_param stderr 1
  procd_close_instance
}

start() {
        service_start $EXEC
	/usr/sbin/swanctl --load-all --noprompt
}

stop() {
        service_stop $EXEC
}

reload() {
        /usr/sbin/swanctl --reload
	/usr/sbin/swanctl --load-all --noprompt
}

```

At least it can do the start and reload job.

stintel, 2020-03-30:
The actual issue reported here should be fixed. Feel free to open a new issue for properly implementing swanctl support.

commit b5eb749a (2020-03-30): strongswan: move ipsec.* to strongswan-ipsec

When building with strongswan-ipsec disabled, strongswan fails to build
because the ipsec.conf file does not exist. Fix this by moving the
ipsec.* files and directories to the strongswan-ipsec package.

Closes #10879 while keeping ipsec.conf to avoid breaking existing
setups, as opposed to #11709.
closed 2020-03-30 by stintel