openwrt/packages #10882: transmission:

ahmar20, 2019-12-27:
Maintainer: @neheb 
Environment:
xrx200, TpLink TD-W8980 OpenWrt v19.07.0-rc2
Description:
I am having a strange issue with transmission. It starts just fine and after a few minutes/hours it starts to give errors as below and stops working and a router reboot fixes it but only for a few minutes. Please note that I do not use IPv6 and the router is connected to main OpenWrt router through LAN and has an IP of 192.168.1.254 while main router has IP of 192.168.1.1.

I dont know why the log shows IPv6 based addresses. Is there something wrong with my config?

config:
```
config transmission
        option enabled '1'
        option config_dir '/tmp/data-hdd/transmission-config'
        option config_overwrite '1'
        option user 'transmission'
        option group 'transmission'
        option mem_percentage '30'
        option nice '10'
        option alt_speed_enabled 'false'
        option bind_address_ipv4 '0.0.0.0'
        option bind_address_ipv6 '::'
        option blocklist_enabled 'false'
        option cache_size_mb '2'
        option dht_enabled 'true'
        option download_dir '/tmp/data-hdd/common-data/torrent-downloads'
        option download_queue_enabled 'true'
        option download_queue_size '1'
        option encryption '1'
        option idle_seeding_limit_enabled 'false'
        option incomplete_dir_enabled 'false'
        option lazy_bitfield_enabled 'true'
        option message_level '1'
        option peer_limit_global '240'
        option peer_limit_per_torrent '60'
        option peer_port '51413'
        option peer_port_random_on_start 'false'
        option peer_socket_tos 'default'
        option pex_enabled 'true'
        option port_forwarding_enabled 'true'
        option preallocation '1'
        option queue_stalled_enabled 'true'
        option queue_stalled_minutes '30'
        option ratio_limit '0.1000'
        option ratio_limit_enabled 'true'
        option rpc_authentication_required 'false'
        option rpc_bind_address '0.0.0.0'
        option rpc_enabled 'true'
        option rpc_host_whitelist '127.0.0.1,192.168.1.*'
        option rpc_host_whitelist_enabled 'false'
        option rpc_port '9091'
        option rpc_url '/transmission/'
        option rpc_whitelist_enabled 'false'
        option scrape_paused_torrents_enabled 'true'
        option script_torrent_done_enabled 'false'
        option speed_limit_down '300'
        option speed_limit_down_enabled 'true'
        option speed_limit_up '20'
        option speed_limit_up_enabled 'true'
        option start_added_torrents 'true'
        option trash_original_torrent_files 'false'
        option umask '18'
        option upload_slots_per_torrent '14'
        option utp_enabled 'true'
        option scrape_paused_torrents 'true'
        option watch_dir_enabled 'false'
        option prefetch_enabled '1'
        option rename_partial_files '1'
        option lpd_enabled 'true'
        option seed_queue_enabled 'true'
        option seed_queue_size '1'
[... 208 lines cut ...]
Thu Dec 26 17:43:53 2019 kern.info kernel: [ 2438.691002] epc = 77f42aa4 in libc.so[77ed0000+98000]
Thu Dec 26 17:43:53 2019 kern.info kernel: [ 2438.696115] ra  = 77e77127 in libevent-2.1.so.7.0.0[77e68000+3a000]
Thu Dec 26 17:44:24 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:44:24.761] Togo (2019) [WEBRip] [1080p] [YTS.LT] Piece 2404, which was just downloaded, failed its checksum test (torrent.c:3327)
Thu Dec 26 17:44:53 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:44:53.757] Togo (2019) [WEBRip] [1080p] [YTS.LT] Piece 1678, which was just downloaded, failed its checksum test (torrent.c:3327)
Thu Dec 26 17:45:17 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:45:17.757] Couldn't connect socket 49 to 2001:ee0:4101:6f90:b9b7:e61e:1438:6b9e, port 35820 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:45:17 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:45:17.758] Couldn't connect socket 49 to 2001:ee0:4101:6f90:b9b7:e61e:1438:6b9e, port 35820 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:45:18 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:45:18.761] Couldn't connect socket 49 to 2600:1702:29f0:3670::566, port 35749 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:45:18 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:45:18.762] Couldn't connect socket 49 to 2600:1702:29f0:3670::566, port 35749 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:46:45 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:46:45.757] Couldn't connect socket 36 to 2409:4071:218a:d471:155d:2ce8:970e:2a98, port 42251 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:47:10 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:47:10.757] Couldn't connect socket 53 to 2a02:8084:90c1:4680:29fa:5012:ed56:e135, port 32239 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:47:10 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:47:10.759] Couldn't connect socket 53 to 2a02:8084:90c1:4680:29fa:5012:ed56:e135, port 32239 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:13 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:13.757] Couldn't connect socket 54 to 2001:8f8:1329:dbd6:fdbf:98fd:af41:9a98, port 36541 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:13 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:13.953] Couldn't connect socket 54 to 2001:8f8:1329:dbd6:fdbf:98fd:af41:9a98, port 36541 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:20 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:20.765] Couldn't connect socket 24 to 2405:204:138e:5639::c6:b8b0, port 43687 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:27 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:27.757] Couldn't connect socket 31 to 2a02:a03f:5634:c700:609c:860f:8761:9957, port 61480 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:27 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:27.759] Couldn't connect socket 31 to 2a02:a03f:5634:c700:609c:860f:8761:9957, port 61480 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:28 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:28.761] Couldn't connect socket 46 to 2a01:e35:2438:b180:2ac6:8eff:fe36:976d, port 51413 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:28 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:28.762] Couldn't connect socket 46 to 2a01:e35:2438:b180:2ac6:8eff:fe36:976d, port 51413 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:40 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:40.757] Couldn't connect socket 56 to 2800:810:533:8441:284d:9dac:d01a:8c98, port 28702 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:40 2019 daemon.err transmission-daemon[1908]: [2019-12-26 17:48:40.759] Couldn't connect socket 56 to 2800:810:533:8441:fc59:b4ab:7928:b484, port 28702 (errno 13 - Permission denied) (net.c:290)
Thu Dec 26 17:48:45 2019 kern.info kernel: [ 2730.257833] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
Thu Dec 26 17:48:45 2019 kern.info kernel: [ 2730.265666] epc = 77f30aa4 in libc.so[77ebe000+98000]
Thu Dec 26 17:48:45 2019 kern.info kernel: [ 2730.270713] ra  = 77e65127 in libevent-2.1.so.7.0.0[77e56000+3a000]
Thu Dec 26 17:48:45 2019 daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 287 seconds since last crash
```

diizzyy, 2019-12-27:
Looks like there's some kind of memory corruption going on, might be the hardware might be a kernel and or compiler issue. Given the hardware I'd recommend you to get a cheap RK3399 board or possibly Allwinner H6 (USB3) and use it for bittorrent as they'll work much better.

ahmar20, 2019-12-27:
I understand that but it works really great on 18.06.5 and there's no error. These errors stared happening a few days ago. Everything was fine before rc2. So I think I should probably wait for this to get stable and then upgrade to it.

neheb, 2019-12-28:
Note that both 18 and 19 use the same version of transmission.

Those IPv6 permission denied errors are known upstream. No fix is available yet.

Interesting that libevent shows up in that kernel report. libevent version varies between the releases.

ahmar20, 2019-12-28:
Could this be related to the fact that dhcp in OpenWrt has started handling IPv6 addresses even though I have disabled IPv6 in `br-lan`. Everything was working fine in v18.06 (because no IPv6) but in v19.07 it's not.

[image]

neheb, 2020-01-01:
Unlikely.

ahmar20, 2020-01-01:
It has started working fairly well on its own. I haven't changed anything in the meantime. My settings already allow only one download and upload at the same time. So memory consumption is within the safe limits.

closed 2020-01-03 by ahmar20