openwrt/packages #10884: apr: Configure contains --without-pic

minfrin, 2019-12-27:
Maintainer @heil

Description:

In the following commit the --without-pic flag has been added to apr. This has memory footprint implications, and ideally should be removed.

https://github.com/openwrt/packages/commit/165f72234dfe8cbb5cfc45b49bf3af134ffa834e

neheb, 2019-12-27:
Increase or reduce?

minfrin, 2019-12-28:
Increases the memory footprint - I understand that a library compiled with position independent code can be loaded and shared amongst processes, while non PIC code has to have a dedicated copy loaded for each process.

On a tight footprint this will make a difference.

neheb, 2019-12-29:
Note that without-pic is passed and $(FPIC) is added. -fPic Vs -fpic

minfrin, 2019-12-30:
I went down an autoconf rabbithole trying to find if a mechanism exists to set -fpic instead of -fPIC, as the current configure script isn't ideal, and I came up with a blank. :(

neheb, 2019-12-30:
I doubt there is one. -fpic vs -fPIC is for slimming down some binaries. -fPIC always works.

micmac1, 2019-12-30:
I agree with @minfrin, there's no good reason for "without-pic". Also `TARGET_CFLAGS += $(FPIC)` should be removed. There's no apparent reason for it.

minfrin, 2019-12-30:
Is the point behind doing ${FPIC} to try and replace -fPIC with -fpic?

micmac1, 2019-12-30:
Nope. What is behind the "$FPIC" (-fPIC) or (-fpic) depends on the target devices.

I've seen this often used in OpenWrt (and I removed it where I could). I think it was used 'a looooong time ago' as a workaround for some build problems, maybe.

micmac1, 2019-12-30:
In some places there is a good reason to add $(FPIC), for instance when upstream doesn't provide a Makefile for a source that is meant to be compiled as a shared object (for instance chan-lantiq, which is a plugin for asterisk). I guess it's possible that some mistook it as a good thing for everything else, too ;)

commit 40a29bf0 (2020-01-12): apr: clean up the package

- removes '$(FPIC)' and '--without-pic', there is no good reason for
  them being used (also @minfrin mentioned memory footprint implications
  when '--without-pic' is used)
- stops removing aclocal.m4 and build/ltmain.sh (reason of removal of
  these files is not apparent anymore)
- adds PKG_BUILD_PARALLEL:=1

Resolves #10884
closed 2020-01-12 by hnyman