openwrt/packages #10885: apr: Upgrade to apr v1.7.1

minfrin, 2019-12-27:
Maintainer @heil

Description:

**Note: apr v1.7.1 has not yet been released, but when it is released**:

- Upgrade to v1.7.1
- Remove 101-fix_apr_time_now.patch, this is now fixed in http://svn.apache.org/viewvc?view=revision&revision=1872035
- Remove 301-apr-fix-off_t-size-doesn-t-match-in-glibc-when-cross.patch, this is  now fixed in http://svn.apache.org/viewvc?rev=1871980&view=rev
- Remove 201-upgrade-and-fix-1.5.1.patch, this is now fixed in http://svn.apache.org/viewvc?rev=1872037&view=rev

micmac1, 2020-02-04:
I think this can be closed. PR is open to upgrade apr to 1.7.0. When 1.7.1 is released we can take care of that, too.

minfrin, 2020-02-05:
I tried to upgrade to apr-1.7.0, and it didn't work me me due to this:

http://mail-archives.apache.org/mod_mbox/apr-dev/202001.mbox/%3C48FD98B5-7417-4EDE-8991-E1088D22B210@sharp.fm%3E

The fix is here:

http://svn.apache.org/viewvc?view=revision&revision=1873594

micmac1, 2020-02-05:
I fixed that with a configure cache. Should be good. Bots are building the packages right now.

micmac1, 2020-02-05:
Mmh, I saw that upstream patch but I don't really like it. Cross-compilers should get the same features as everybody else.

minfrin, 2020-02-05:
I believe that's what the patch does - there is a broken autoconf test that detects the wrong capability, but then we try to use the correct feature, and the compile fails as the prerequisites are not there. This has been fixed.

micmac1, 2020-02-05:
The fix from upstream looks OK, but it doesn't allow the use of the new apr_lock strategy when cross-compiling. For that I have introduced a patch which allows a cached answer for the mmap /dev/zero check, so we can use the new apr_lock strategy as well. So all is well.

minfrin, 2020-02-05:
If you introduce patches, please bring the discussion and the patch to the dev@apr.apache.org mailing list so that it can be considered upstream.

If there are problems using any part of APR we want to know about them, so that everyone can benefit.

micmac1, 2020-02-05:
I think I raised a bug with an added cache check for strerror to Apache a year or two ago, but there was no response. You think mailing list makes more sense?

minfrin, 2020-02-05:
Certainly ping the mailing list to get the bug more attention. Bugzilla can be a hole to get lost in some times. :(

micmac1, 2020-02-18:
I sent the patches to the mailing list a few days ago. This issue can be closed now, right?

neheb, 2020-07-23:
Might as well.

closed 2020-07-23 by neheb