openwrt/packages #10899: motion fails to start

jaro3, 2019-12-29:
cannot get this to start either streaming or webcontrol interface, no netstat binding to any port, this is not a tls neither localhost issue....

motion - 4.2.2-1

motion_startup: Motion running as daemon process
webu_start_strm: Starting all camera streams on port 8081
webu_start_strm: Unable to start stream for camera 0
webu_start_ctrl: Starting webcontrol on port 8080
webu_start_ctrl: Unable to start MHD

neheb, 2019-12-29:
Is this on 19.07?

jaro3, 2019-12-29:
trunk git-19.291.72323  r11254

neheb, 2019-12-29:
Update your feeds. This was fixed.

jaro3, 2019-12-29:
Thx, could you point me to the fix/patch ?

neheb, 2019-12-29:
https://github.com/openwrt/packages/tree/master/multimedia/motion/patches

jaro3, 2019-12-31:
works, fantastic!

closed 2019-12-31 by BKPepe