openwrt/packages #10911: smartd: include /etc/smartd_warning.sh script for sending emails

gtbX, 2019-12-31:
Maintainer: @mstorchak 
Environment: ARMv7, Linksys WRT1200AC, OpenWrt 18.06.5 r7897-9d401013fc

Description:
The `smartd` package doesn't install the /etc/smartd_warning.sh script or the /etc/smartd_warning.d/ plugin directory. This script is used by smartd to format failure messages when it is configured to email results. I was able to get it working by copying the file from another installation.

mstorchak, 2020-01-01:
By default the script doesn't look very useful: it relies on `mail` command for sending email, but AFAIK none of the packages provides `mail` interface for mail sending.
Could you please share your config or explain how it works for you?

gtbX, 2020-01-01:
I installed the mailx binary from the 'nail' package, and configured smartd to use it by adding to smartd.conf:
```
DEFAULT -M exec /usr/bin/mailx ...
```
I then used msmtp and msmtp-mta as the actual sendmail implementation to email smart warnings out.

mstorchak, 2020-01-02:
Thank you for the information. It looks the script should be a packaged separately, and this package should depend on smartd, nail and msmtp-mta.

gtbX, 2020-01-02:
A separate package should be OK (I get trying to keep packages small).
I don't know if any other packages provide a `mail` implementation, but ssmtp provides a `sendmail` implementation that should work as well.

closed 2020-01-06 by hnyman