openwrt/packages #10919: transmission-daemon-openssl: message_level setting not work

Mosney, 2020-01-01:
Maintainer: @neheb @ysc3839 @ecsv @korzhyk @hnyman
Environment: 
**OpenWrt x86-64 19.07rc2**
Some detail below:
```
root@OP19:~# ps |grep transmission |grep -v grep
23540 transmis 54580 SN   /usr/bin/transmission-daemon -f --log-error -g /etc/transmission
root@OP19:~# cat /etc/config/transmission |grep message
        option message_level '3'
root@OP19:~# cat /etc/transmission/settings.json |grep message
    "message-level": 1,
```
```
root@OP19:~# opkg list-installed |grep transmission
luci-app-transmission - git-19.365.55415-a44e6b3-1
luci-i18n-transmission-zh-cn - git-19.365.55415-a44e6b3-1
transmission-daemon-openssl - 2.94-9
transmission-web-control - 2019-04-16-a747a87b-1
```
Description:
I noticed init.d script use `--log-error`, maybe it is the reason? 
https://github.com/openwrt/packages/blob/5a25cf86ea56d2019ef551b727918c2e33427502/net/transmission/files/transmission.init#L117

Additionally, I tried to change transmission config file `message level 0` directly and also set luci setting `option config_overwrite '0'` to confirm init script will not modify it. But in the end transmission still output log and `message level 0` still has been modified to `message level 1`. Even I try to review init script but I don't know why.

neheb, 2020-01-01:
--log-error overrides it. Looks like I'll have to remove it.

Mosney, 2020-01-01:
https://github.com/openwrt/luci/issues/3432#issuecomment-568902001
Looks I'm right, and I'll try to delete this part to run for test.

Mosney, 2020-01-01:
https://github.com/openwrt/packages/blob/5a25cf86ea56d2019ef551b727918c2e33427502/net/transmission/files/transmission.init#L117

I tried direct delete `--log-error`, It worked. Hoping for a new fixed version.

neheb, 2020-01-01:
Can you try using this init script? https://gist.github.com/neheb/a160d973643a208f54cdcfbe8d7c238a

I made several cleanups.

neheb, 2020-01-01:
Bah stupid mistake here: https://gist.github.com/neheb/a160d973643a208f54cdcfbe8d7c238a  . Must be terminated with ]

Mosney, 2020-01-01:
Cannot run, no syslog message.

neheb, 2020-01-01:
Try this: https://gist.github.com/neheb/91e1fc794864f526ed71e17f730b17b4

Mosney, 2020-01-01:
Looks good for now

closed 2020-01-12 by neheb