openwrt/packages #10922: <ddns-scripts> ddns-scripts_cloudflare.com-v4 - subdomain problem and missing docs

aleqx, 2020-01-01:
I eventually was able to fix it by reverse engineering the code, but it took me frustratingly long, so I'd like to suggest a better (proper) implementation be made and some actual docs. Currently there are no docs on this and the interface is misleading, causing errors. The only bit of helpful info is a rather obscure post within a bug report thread in here https://github.com/openwrt/packages/issues/5097

- Let's please fix the non-standard and undocumented `subdomain@domain.com` format and instead allow the expected and intuitive `subdomain.domain.com` like all others. I'm aware the core ddns script is splitting the fqdn but using `@` is a very dirty workaround. Why not instead check if the ddns service is cloudflare-v4 and do the needful cleanly? The dirty `@` can still be kept for backward compatibility (even if ugly) to prevent breaking people's current configurations.

Obviously I first tried `subdomain.domain.com` which actually does not fail! It changes the IP for `domain.com` (instead of `subdomain.domain.com`) which redirected my website and all services using `doomain.com` to my home IP. This isn't just annoying, it's an actual problem (think about an actual business website!).

- Could I also suggest writing some actual docs for how to configure ddns with cloudflare-v4 for openwrt (both config files and LUCI GUI)?

For others (like me) who struggled, my working /etc/config/ddns is:
```
config ddns 'global'
        option ddns_dateformat '%F %R'
        option ddns_loglines '250'
        option upd_privateip '0'

config service 'box_ddns'
        option service_name 'cloudflare.com-v4'
        option ip_source 'web'
        option ip_url 'http://icanhazip.com'
        option lookup_host 'box.domain.com'
        option username '<your_cloudflare_email_login>'
        option password '<your_cloudflare_api_key>'
        option enabled '1'
        option use_https '1'
        option domain 'box@domain.com'
```
Note the `box@domain.com` instead of `box.domain.com` for the `domain` entry (the latter is what you actually want) and `box.domain.com` for the `lookup_host` entry. Quite confusing and non-intuitive.

If you use the LUCI DDNS GUI then use:

Lookup Hostname: box.domain.com
Domain: box@domain.com
Username: <your_cloudflare_email_login>
Password: <your_cloudflare_api_key>

h4waii, 2020-01-01:
I have to agree. This really needs to be documented somewhere.

Also to note, you must use the **Global API**, not a generated API token with scoped access.

zzzxxx3221421, 2020-02-25:
it is hard to understand

edrikk, 2022-01-16:
I had to work through multiple issues, and reach the shell script to get this working.

The first stumbling block was indeed the “@“ separator requirement noted above.

Once passed that, I kept getting the error:
`Could not route to /zones/dns_records, perhaps your object identifier is invalid?"}, `

looking through the code, there is an it statement whereby if the zone_id is not given, it tried to figure it out…. But either there’s a bug in the logic, or somehow my error.  But regardless, adding zone_id value to the ddns config file immediately resolved the issue.

Don’t know why zone_id isn’t an optional parameter in the GUI.  Could save people hours.

edrikk, 2022-01-16:
Looks like a fix was just put in for the zine_id if statement issue:

https://github.com/openwrt/packages/pull/17617