openwrt/packages #10932: php7-mod-gd: module missing dependencies?

flyn-org, 2020-01-03:
Maintainer: @mhei 
Environment: master as of 1/2/20.

Description:
It seems that the php7-mod-gd 7.4.1 or perhaps libgd is missing a dependency on libwebp. I ran the following on an x86_64 OpenWrt host.

# ldd /usr/lib/php/gd.so 
	ldd (0x7ffb364b8000)
	libgd.so.3 => /usr/lib/libgd.so.3 (0x7ffb36451000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7ffb3643d000)
	libc.so => ldd (0x7ffb364b8000)
	libz.so.1 => /usr/lib/libz.so.1 (0x7ffb36429000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x7ffb363fb000)
	libjpeg.so.9 => /usr/lib/libjpeg.so.9 (0x7ffb363c8000)
Error loading shared library libwebp.so.7: No such file or directory (needed by /usr/lib/libgd.so.3)
[...]
# opkg install libwebp
Installing libwebp (1.0.3-2) to root...

neheb, 2020-01-03:
Does ldd throw errors with libgd or libwebp?

flyn-org, 2020-01-04:
The ldd utility throws errors with libgd. However, something strange is going on, because when I reinstalled libgd it pulled in libwebm. I do see libwebm explicitly listed in libgd's package Makefile. I will take another look after a fresh install---I don't know why libgd failed to pull in libwebp before.

# ldd ./usr/lib/libgd.so.3.0.5
	ldd (0x7f7fefaaf000)
	libz.so.1 => /usr/lib/libz.so.1 (0x7f7fefa4e000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x7f7fefa20000)
	libjpeg.so.9 => /usr/lib/libjpeg.so.9 (0x7f7fef9ed000)
Error loading shared library libwebp.so.7: No such file or directory (needed by ./usr/lib/libgd.so.3.0.5)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f7fef9d9000)
	libc.so => ldd (0x7f7fefaaf000)

mhei, 2020-01-12:
Assigning to myself, to keep it on my radar...

flyn-org, 2020-01-14:
Not sure what I did wrong before reporting this, but a fresh build and install of current master did not exhibit this problem. I will reopen with more information if I see this again.

closed 2020-01-14 by flyn-org