openwrt/packages #10942: nfs-kernel-server broken

Chris56, 2020-01-04:
Maintainer: @\tripolar

Environment: 18.06.*, all models and archs

Description:
Vanilla fresh install of 18.06.5
opkg install nfs-kernel-server
You immediately see:
```
exportfs: /mnt does not support NFS export
```
And immerdiately a flood of kernel/daemon errors/warning
```
Sat Jan  4 06:34:49 2020 kern.warn kernel: [   92.398917] svc: failed to register nfsdv3 RPC service (errno 124).
Sat Jan  4 06:34:50 2020 kern.warn kernel: [   92.490088] svc: failed to register lockdv1 RPC service (errno 124).
Sat Jan  4 06:34:50 2020 kern.warn kernel: [   92.503551] NFSD: the nfsdcld client tracking upcall will be removed in 3.10. Please transition to using nfsdcltrack.
Sat Jan  4 06:34:50 2020 kern.info kernel: [   92.524731] NFSD: starting 90-second grace period (net 8054f5e0)
Sat Jan  4 06:34:50 2020 daemon.notice rpc.statd[9399]: Version 2.3.3 starting
Sat Jan  4 06:34:50 2020 daemon.warn rpc.statd[9399]: Flags: No-Daemon TI-RPC
Sat Jan  4 06:34:50 2020 daemon.notice rpc.statd[9399]: Initializing NSM state
Sat Jan  4 06:34:50 2020 daemon.err rpc.mountd[9400]: Could not bind socket: (125) Address in use
Sat Jan  4 06:34:50 2020 daemon.warn rpc.statd[9399]: Running as root.  chown /var/lib/nfs to choose different user
Sat Jan  4 06:34:50 2020 daemon.err rpc.mountd[9400]: Could not bind socket: (125) Address in use
Sat Jan  4 06:34:50 2020 daemon.err rpc.mountd[9400]: Could not bind socket: (125) Address in use
Sat Jan  4 06:34:50 2020 daemon.err rpc.statd[9399]: Could not bind socket: (125) Address in use
Sat Jan  4 06:34:50 2020 daemon.err rpc.mountd[9400]: Could not bind socket: (125) Address in use
Sat Jan  4 06:34:50 2020 daemon.err rpc.mountd[9400]: Could not bind socket: (125) Address in use
Sat Jan  4 06:34:50 2020 daemon.warn rpc.mountd[9400]: mountd: No V2 or V3 listeners created!
Sat Jan  4 06:34:50 2020 daemon.notice rpc.mountd[9400]: Version 2.3.3 starting
Sat Jan  4 06:34:50 2020 daemon.err rpc.statd[9399]: failed to create RPC listeners, exiting
```
Thus, nfs3 does not work anymore, which kills all former setups as they had to be v3 because v4 wasn't there!
This bug has been there for more than ONE YEAR!
Aren't packages tested at all before a release? Not even a simple install?
I'm not talking about a subtle complex situation, when some intricate problem shows up!!
Just a plain vanilla installation. Everything wiped and just a plain "install nfs-kernel-server" !!!!

BKPepe, 2020-01-04:
Duplicate of #8930

closed 2020-01-04 by BKPepe

ptpt52, 2020-01-09:
but something wrong in menuconfig
```
make menuconfig
tmp/.config-package.in:107474:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:107474:	symbol PACKAGE_nfs-kernel-server depends on NFS_KERNEL_SERVER_V4
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
feeds/packages/net/nfs-kernel-server/Config.in:4:	symbol NFS_KERNEL_SERVER_V4 depends on PACKAGE_nfs-kernel-server
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
```

@BKPepe @Chris56