openwrt/packages #10944: php7: does not require libgd to build with freetype support

flyn-org, 2020-01-04:
Maintainer: @mhei 
Environment: x86_64 master

Description:
Commit 0f10c8c8 caused php7-mod-gd to build against the external libgd library. However, the libgd library does not build by default with freetype support. This leaves php7-mod-gd lacking symbols---such as gdFontCacheShutdown---needed by many PHP programs. Here is an example:

```
# php-cli occ upgrade
PHP Warning:  PHP Startup: Unable to load dynamic library 'gd.so' (tried: /usr/lib/php/gd.so (Error relocating /usr/lib/php/gd.so: gdFontCacheShutdown: symbol not found), /usr/lib/php/gd.so.so (Error loading shared library /usr/lib/php/gd.so.so: No such file or directory)) in Unknown on line 0
The process control (PCNTL) extensions are required in case you want to interrupt long running commands - see http://php.net/manual/en/book.pcntl.php
The current PHP memory limit is below the recommended value of 512MB.
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
PHP module GD not installed.
Please ask your server administrator to install the module.

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /mnt/xvdb/var/www/flyn.org/share-new/lib/private/Console/Application.php:167
Stack trace:
#0 /mnt/xvdb/var/www/flyn.org/share-new/console.php(98): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /mnt/xvdb/var/www/flyn.org/share-new/occ(11): require_once('/mnt/xvdb/var/w...')
```

Is it possible that php7-mod-gd could require that libgd build with freetype support? Or, perhaps libgd should build with freetype support by default. Note that I have filled a separate pull #10943   which describes why building libgd with freetype presently appears broken.

mhei, 2020-01-05:
I stumbled over this problem a few days ago, too. But due to xmas holidays, I did not found enough time to dig into this.
Points I would like to clarify would be e.g.:

- is it possible for libgd client to detect whether libgd is compiled with freetype support?
- if so, does the php gd extension makes use of such a feature?
- beside the points above, I agree that a users of php-mod-gd probably almost want to have freetype support and do not really care about package size etc. but if we declare a dependency to libgd's freetype support, then the buildbots would build it always with it enabled which might conflict with other libgd users (I did not check libgd for other users though)

flyn-org, 2020-01-05:
From what I can tell so far, PHP does not try to determine whether libgd provides freetype support. I was a little surprised that PHP built without it, but I suppose that is because the output of the PHP build is gd.so, a shared object. The linker will not complain about gd.so's unresolved symbols.

The following packages seem to depend (sometimes optionally) on libgd:

multimedia/fswebcam
libs/serdisplib
lang/php7
net/vnstat
net/apcupsd
net/nut
utils/bandwidthd
utils/lcd4linux

My first instinct is to bite the bullet and build freetype support into libgd by default.

We could patch PHP to pull out the font stuff from the gd module, but that would probably surprise someone who runs a PHP application that requires it.

What about modifying libgd to follow the vim package's vim and vim-full model? I think ffmpeg/libffmpeg might do this too. It looks like the build system builds all of the variants in these cases.

flyn-org, 2020-01-05:
@jow, any comments?

mhei, 2020-01-12:
Fixed with fa0d982c77200910f5fe2061531580ca3f362b1a

closed 2020-01-12 by mhei