openwrt/packages #10961: openvswitch: [19.07] build fails woth "too few arguments to function 'skb_dst"

SvenRoederer, 2020-01-07:
Maintainer: @yousong
Environment: (OpenWrt-19.07 release-tag)

Description: We build our firmware with CONFIG_KMOD_ALL enabled and the build fails with the following error:

```
  CC [M]  firmware/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/openvswitch-2.11.0/datapath/linux/ip_tunnel.o
firmware/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/openvswitch-2.11.0/datapath/linux/ip_tunnel.c: In function 'rpl_tnl_update_pmtu':
firmware/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/openvswitch-2.11.0/datapath/linux/ip_tunnel.c:269:3: error: too few arguments to function 'skb_dst((const struct sk_buff *)skb)->ops->update_pmtu'
   skb_dst(skb)->ops->update_pmtu(skb_dst(skb), NULL, skb, mtu);
   ^~~~~~~
make[8]: *** [scripts/Makefile.build:327: firmware/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/openvswitch-2.11.0/datapath/linux/ip_tunnel.o] Error 1

```

yousong, 2020-01-08:
The issue should now be fixed in both master and openwrt-19.07 branch.  Thanks for the heads-up

SvenRoederer, 2020-01-08:
Thanks, it's fixed for 19.07 with 74e160df2218399c1372f2c51b5bdc77747c7cb0. So I assume 7cc9c437beb798be1699abb403e8964a43252939 will do on master also.

closed 2020-01-08 by SvenRoederer