openwrt/packages #10965: daemon.err uwsgi

ferre-han, 2020-01-08:
Environment: (OpenWrt / LuCI Master git-20.007.29419-38c9c9e)

Description:

```
Wed Jan  8 11:14:39 2020 daemon.err uwsgi[3666]: *** has_emperor mode detected (fd: 7) ***
Wed Jan  8 11:14:39 2020 daemon.err uwsgi[3666]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-webui.ini
Wed Jan  8 11:14:39 2020 daemon.err uwsgi[3666]: Wed Jan  8 11:14:39 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini has been spawned
Wed Jan  8 11:14:39 2020 daemon.err uwsgi[3666]: Wed Jan  8 11:14:39 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Wed Jan  8 11:14:40 2020 daemon.err uwsgi[3666]: Wed Jan  8 11:14:40 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Wed Jan  8 11:14:45 2020 daemon.info uwsgi-luci: luci: accepted login on / for root from 192.168.2.128
Wed Jan  8 11:14:49 2020 daemon.err uwsgi[3666]: Wed Jan  8 11:14:49 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Wed Jan  8 11:14:52 2020 daemon.err uwsgi[3666]: *** has_emperor mode detected (fd: 8) ***
Wed Jan  8 11:14:52 2020 daemon.err uwsgi[3666]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-cgi_io.ini
Wed Jan  8 11:14:52 2020 daemon.err uwsgi[3666]: Wed Jan  8 11:14:52 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini has been spawned
Wed Jan  8 11:14:52 2020 daemon.err uwsgi[3666]: Wed Jan  8 11:14:52 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-cgi_io.ini is ready to accept requests```

neheb, 2020-01-12:
I don't see the issue here.

closed 2020-01-14 by ferre-han

cconde, 2020-10-21:
Same here, there is a lot of error logs that either are info or they are error to be resolved.
In the first case they should be moved to daemon info, in the second case there should be a description of what error is being reported at daemon.err

neheb, 2020-10-21:
https://github.com/openwrt/packages/blob/master/net/uwsgi/files/emperor.ini could be modified to add log-syslog. I have no idea what that will log to.

That and switching https://github.com/openwrt/packages/blob/master/net/uwsgi/files/uwsgi.init#L9 to 0.