openwrt/packages #10967: golang: Illegal Instruction when arm cpu do not have VFP

kuoruan, 2020-01-08:
Maintainer: @jefferyto
Environment: armv7

Description:

Some CPU do not have VFP.

For example: Phicomm K3

the default GOARM for K3 is 7, but the build would case Illegal Instruction. (golang/go#18483)

to make it work, we need to set GOARM to 5.

```
root@Phicomm-K3:~# cat /proc/cpuinfo
processor : 0
model name : ARMv7 Processor rev 0 (v7l)
BogoMIPS : 1400.00
Features : half thumb fastmult edsp tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x3
CPU part : 0xc09
CPU revision : 0

processor : 1
model name : ARMv7 Processor rev 0 (v7l)
BogoMIPS : 1400.00
Features : half thumb fastmult edsp tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x3
CPU part : 0xc09
CPU revision : 0

Hardware : BCM5301X
Revision : 0000
Serial : 0000000000000000
```

jefferyto, 2020-01-09:
Thanks for reporting this - I think I've figured out how to address this, but I will wait until #10971 is done to open a PR for this fix.

jefferyto, 2020-01-14:
Fixes are in #11020 (master), #11021 (19.07), and #11023 (18.06)

closed 2020-01-14 by neheb

kuoruan, 2020-01-15:
Great! Thank you.

PR #11023 (2020-01-14): [openwrt-18.06] golang: Fix selection of GOARM value
Maintainer: me
Compile tested: at91-legacy / brcm2708-bcm2708 / mvebu-cortexa9 / at91-sama5d2 / imx6-generic / armvirt-32, 18.06.6 release sdk
Run tested: none

Description:
This fixes how `GOARM` is selected for arm platforms, based on support for VFP/VFPv3 rather than CPU version.

Fixes #10967.
commit 4c6ac36d (2020-01-14): golang: Fix selection of GOARM value

This fixes how GOARM is selected for arm platforms, based on support for
VFP/VFPv3 rather than CPU version.

Fixes #10967.
commit b36f2771 (2020-01-14): golang: Fix selection of GOARM value

This fixes how GOARM is selected for arm platforms, based on support for
VFP/VFPv3 rather than CPU version.

Fixes #10967.
commit fdd202bd (2020-01-14): golang: Fix selection of GOARM value

This fixes how GOARM is selected for arm platforms, based on support for
VFP/VFPv3 rather than CPU version.

Fixes #10967.