openwrt/packages #10977: i2pd: does not start with init script

BKPepe, 2020-01-10:
Maintainer: @yangfl 
Environment: Turris Omnia, mvebu (cortex-a9), OpenWrt master

Description:

```
Installing i2pd (2.28.0-1.6) to root...
Configuring i2pd.
root@turris:~# /etc/init.d/i2pd start
chown: : No such file or directory
```

I can start it without init script just fine.

neheb, 2020-01-12:
I assume the pidfile functionality can be removed since this is using procd.

closed 2020-01-17 by neheb

BKPepe, 2020-01-31:
Same issue in OpenWrt 19.07.

```
root@turris:/# /etc/init.d/i2pd start
mkdir: can't create directory '': No such file or directory
chown: : No such file or directory
chown: : No such file or directory
```

neheb, 2020-01-31:
Would be good to backport that commit.

amzamzamzamz, 2021-07-25:
> Same issue in OpenWrt 19.07.
> 
> ```
> root@turris:/# /etc/init.d/i2pd start
> mkdir: can't create directory '': No such file or directory
> chown: : No such file or directory
> chown: : No such file or directory
> ```

Change i2pd in /etc/init.d/ to:

`#!/bin/sh /etc/rc.common
# Copyright (C) 2017 OpenWrt.org

USE_PROCD=1

START=90
STOP=10

# default params
PROG=/usr/sbin/i2pd
USER="i2pd"
GROUP="i2pd"
PIDFILE=/var/run/i2pd.pid
DATADIR=/var/lib/i2pd
CONFFILE=/etc/i2pd/i2pd.conf
ADDRESSBOOKDIR=/root/.i2pd/addressbook

. /lib/functions.sh

start_service() {
	local data_dir
	local addressbook_dir

	config_load i2pd

#	config_get data_dir i2pd data_dir
#	config_get addressbook_dir i2pd addressbook_dir

	## Setting up data dir
	if [ ! -d "$DATADIR" ]; then
		mkdir -p "$DATADIR"
		ln -s /usr/share/i2pd/certificates "$DATADIR/certificates"
		ln -s /etc/i2pd/tunnels.conf "$DATADIR/tunnels.conf"
		if [ -n "$ADDRESSBOOKDIR" ]; then
			if [ ! -d "$ADDRESSBOOKDIR" ]; then
				mkdir -p "$ADDRESSBOOKDIR"
			fi
			ln -s "$ADDRESSBOOKDIR" "$$DATADIR/addressbook"
		fi
	fi

	## We need permissions
	chown "$USER:$GROUP" "$DATADIR"
	chown "$USER:$GROUP" "$ADDRESSBOOKDIR"
	touch "$PIDFILE"
	chown "$USER:adm" "$PIDFILE"

	procd_open_instance
	procd_set_param command "$PROG" --service --conf="$CONFFILE" --pidfile "$PIDFILE" --http.enabled 1 --http.address 127.0.0.1 --http.port 7070  --http.auth 0
	## Don't know about i2pd user's HOME
	procd_set_param env "HOME=$DATADIR"
	procd_set_param limits nofile=4096
	procd_set_param stdout 1
	procd_set_param stderr 1
	procd_set_param user "$USER"
	procd_set_param pidfile "$PIDFILE"
	procd_close_instance
}`

There is no errors at service starting but it can't connect to any i2p-router. So it doesn't work anyway