openwrt/packages #10982: meson packages: Unknown compiler(s)

BKPepe, 2020-01-11:
Maintainer: @dhewg 
Tested environment: Turris Omnia, mvebu, OpenWrt 19.07 and OpenWrt master

Description:

I can not compile libsoup even Knot Resolver (not in this feed). Maybe more packages, but these are which I tested until I add these two rows in their Makefiles:
```
HOSTCC=$(HOSTCC_NOCACHE)
HOSTCXX=$(HOSTCXX_NOCACHE)
```

Otherwise it fails:
```
The Meson build system
Version: 0.52.1
Source dir: /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libsoup-2.68.2
Build dir: /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/libsoup-2.68.2/openwrt-build
Build type: cross build
Project name: libsoup
Project version: 2.68.2

meson.build:1:0: ERROR: Unknown compiler(s): [['ccache gcc']]
The follow exceptions were encountered:
Running "ccache gcc --version" gave "[Errno 2] No such file or directory: 'ccache gcc': 'ccache gcc'"
```

```
The Meson build system
Version: 0.52.1
Source dir: /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/knot-resolver-4.3.0
Build dir: /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/knot-resolver-4.3.0/openwrt-build
Build type: cross build
Project name: knot-resolver
Project version: 4.3.0

meson.build:1:0: ERROR: Unknown compiler(s): [['ccache gcc']]
The follow exceptions were encountered:
Running "ccache gcc --version" gave "[Errno 2] No such file or directory: 'ccache gcc': 'ccache gcc'"

A full log can be found at /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/knot-resolver-4.3.0/openwrt-build/meson-logs/meson-log.txt
```

### Contents of meson-log.txt:
```
Build started at 2020-01-11T13:38:37.702423
Main binary: /**path**/build/staging_dir/host/bin/python
Build Options: -Dprefix=/usr -Dbuildtype=plain '--cross-file /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/knot-resolver-4.3.0/openwrt-cross.txt' '--native-file /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/knot-resolver-4.3.0/openwrt-native.txt'
Python system: Linux
The Meson build system
Version: 0.52.1
Source dir: /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/knot-resolver-4.3.0
Build dir: /**path**/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/knot-resolver-4.3.0/openwrt-build
Build type: cross build
Project name: knot-resolver
Project version: 4.3.0

meson.build:1:0: ERROR: Unknown compiler(s): [['ccache gcc']]
The follow exceptions were encountered:
Running "ccache gcc --version" gave "[Errno 2] No such file or directory: 'ccache gcc': 'ccache gcc'"
```

dhewg, 2020-01-11:
^ that should fix it

BKPepe, 2020-01-11:
I can confirm it. It works.

PR #10984 (2020-01-11): meson: fix building host binaries with ccache
Maintainer: me
Compile tested: sunxi, master
Run tested: -

Description:
With CONFIG_CCACHE, $HOSTCC is 'ccache gcc' while $TARGET_CC points to a
single wrapper script. Compiling target binaries with ccache works, but
doesn't for host binaries, because we need to supply an argv array for
argc > 1 in the meson cross file.

Always pass an array for the c and c++ compiler, and while at it, do it
for the target as well - just to be on the safe side if that ever
changes.

Fixes #10982.
commit f5c8565d (2020-01-11): meson: fix building host binaries with ccache

With CONFIG_CCACHE, $HOSTCC is 'ccache gcc' while $TARGET_CC points to a
single wrapper script. Compiling target binaries with ccache works, but
doesn't for host binaries, because we need to supply an argv array for
argc > 1 in the meson cross file.

Always pass an array for the c and c++ compiler, and while at it, do it
for the target as well - just to be on the safe side if that ever
changes.

Fixes #10982.
commit c7e1a345 (2020-04-06): meson: fix building host binaries with ccache

With CONFIG_CCACHE, $HOSTCC is 'ccache gcc' while $TARGET_CC points to a
single wrapper script. Compiling target binaries with ccache works, but
doesn't for host binaries, because we need to supply an argv array for
argc > 1 in the meson cross file.

Always pass an array for the c and c++ compiler, and while at it, do it
for the target as well - just to be on the safe side if that ever
changes.

Fixes #10982.

(cherry picked from commit f5c8565d1bbd8f97b1f62319398543a56d591814)
closed 2020-01-11 by neheb