openwrt/packages #10983: canutils: can not be compiled with ASLR PIE

BKPepe, 2020-01-11:
Maintainer: @toxxin 
Environment: OpenWrt master, mvebu, Turris Omnia

Description:

PKG_ASLR_PIE is disabled by default and when I enabled it, I can not compile canutils:

```
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: candump.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
candump.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'candump' failed
make[3]: *** [candump] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   cansend.o lib.o   -o cansend
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: cansend.o: relocation R_ARM_MOVW_ABS_NC against `stderr' can not be used when making a shared object; recompile with -fPIC
cansend.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'cansend' failed
make[3]: *** [cansend] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   canplayer.o lib.o   -o canplayer
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: canplayer.o: relocation R_ARM_MOVW_ABS_NC against `stderr' can not be used when making a shared object; recompile with -fPIC
canplayer.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'canplayer' failed
make[3]: *** [canplayer] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   cangen.o lib.o   -o cangen
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: cangen.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
cangen.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'cangen' failed
make[3]: *** [cangen] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   canbusload.o canframelen.o   -o canbusload
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: canbusload.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
canbusload.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'canbusload' failed
make[3]: *** [canbusload] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   log2long.o lib.o   -o log2long
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: log2long.o: relocation R_ARM_MOVW_ABS_NC against `stdin' can not be used when making a shared object; recompile with -fPIC
log2long.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'log2long' failed
make[3]: *** [log2long] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   log2asc.o lib.o   -o log2asc
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: log2asc.o: relocation R_ARM_MOVW_ABS_NC against `stderr' can not be used when making a shared object; recompile with -fPIC
log2asc.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'log2asc' failed
make[3]: *** [log2asc] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   asc2log.o lib.o   -o asc2log
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: asc2log.o: relocation R_ARM_MOVW_ABS_NC against `stderr' can not be used when making a shared object; recompile with -fPIC
asc2log.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'asc2log' failed
make[3]: *** [asc2log] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   canlogserver.o lib.o   -o canlogserver
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: canlogserver.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
canlogserver.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'canlogserver' failed
make[3]: *** [canlogserver] Error 1
ccache_cc -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/lib -L/**path**/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/lib -L/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib -fpic -specs=/**path**/build/include/hardened-ld-pie.specs -znow -zrelro   jacd.o libj1939.o   -o jacd
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: jacd.o: relocation R_ARM_MOVW_ABS_NC agains [... cut ...]
/**path**/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.5.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: testj1939.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
testj1939.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'testj1939' failed
make[3]: *** [testj1939] Error 1
make[3]: Target 'all' not remade because of errors.
```

neheb, 2020-01-12:
Please try this patch: https://gist.github.com/neheb/714676c6b926b8f3d40434ddec5d04d1

The FLAGS are not being passed at all.

BKPepe, 2020-01-12:
Suggested changes in your patch works. I have been able to compile it and to be sure, I briefly run tested too. It seems to be OK.

closed 2020-01-12 by neheb