openwrt/packages #10985: zip package missing in 19.07.0.

4k3or3et, 2020-01-11:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```
I would like report a bug where package zip is missing 19.07.0. While installing i get an output:

`Unknown package 'zip'.
Collected errors:
 * opkg_install_cmd: Cannot install package zip.`

Thank you.

neheb, 2020-01-11:
It should be available. Which platform?

Note that you need to run opkg update first.

4k3or3et, 2020-01-11:
> It should be available. Which platform?
> 
> Note that you need to run opkg update first.

opkg update command had been ran before trying to install zip package. 

platform is ramips - mt7621. Router ZBT WE1326.

hnyman, 2020-01-11:
Copied my response from forum:

Likely a bug.
I can't install it on master although the package Makefile is in the packages feed repo and looks normal at the first glance.

The peculiar part is that there is  no downloadable package, and no item in the package listing, but still there is no failure log at buildbot.

But I was able to compile it by myself and install that .ipk in my router quite normally. Strange.

18.06 has the buildbot downloadable package, but master and 19.07 do not.

4k3or3et, 2020-01-11:
Sorry guys but i am noob here. Is there anyone who can fix that?

hnyman, 2020-01-11:
I believe that I figured out the reason: name conflict

@mkresin introduced zip as a main repo build tool a year ago with https://github.com/openwrt/openwrt/commit/ad8c2d6099a7da5192ee6e2c52a8303642271877
That is now both in 19.07 and master.

And that host tool is defined as a package, so it likely prevents the actual installable package from the packages feed to get compiled in buildbot.

hnyman, 2020-01-11:
Well, there is not actual package definition in the host tool, so it is likely not a full package, and should not interfere.

But there is a commented out last line, the buildpackage eval call. Could that interfere in buildbot, even if there is no target package definition?

neheb, 2020-01-15:
ping @jow-

4k3or3et, 2020-01-16:
Is there any other way i could encourage someone who can fix this? Thank you.

hnyman, 2020-01-16:
> Is there any other way i could encourage someone who can fix this? Thank you.

You can build it by yourself.

It may be something in the buildbot specific package evaluation logic that prevents the package being compiled there.

4k3or3et, 2020-01-16:
> > Is there any other way i could encourage someone who can fix this? Thank you.
> 
> You can build it by yourself.
> 
> It may be something in the buildbot specific package evaluation logic that prevents the package being compiled there.

I am complete noob in Linux environment. Would you be that kind and tell me how do i start?

hnyman, 2020-01-16:
https://openwrt.org/docs/guide-developer/build-system/start

diizzyy, 2020-01-16:
Use bsdtar instead (libarchive)? :-)

4k3or3et, 2020-01-17:
> https://openwrt.org/docs/guide-developer/build-system/start

I thought it is about compiling only zip package from the source. It looks like the matter is to create entire OpenWRT image. I will be honest, that is bigger than me.

Do you think guys someone will eventually fix the missing zip package or that is just my wishful thinking?

4k3or3et, 2020-01-17:
> bsdtar

Thank you but the it has to be really zip as it has to work natively on Windows without any additional decompressing software...

diizzyy, 2020-01-18:
bsdtar generates zip-files compatible with Windows
https://www.libarchive.org/

neheb, 2020-01-22:
@hnyman That's probably accurate.

utils/zip has

```
$(eval $(call HostBuild))
$(eval $(call BuildPackage,zip))
```

tools/zip has

```
$(eval $(call HostBuild))
#$(eval $(call BuildPackage,zip))
```

I think the HostBuild is conflicting here. OTOH I have this commit:

https://github.com/openwrt/packages/commit/b61ca9d885fc323d6d3f0a5a08a6fe09e20e9ad2

I will make a draft PR to test this.

diizzyy, 2020-01-22:
...or take the opportunity to get rid of something that hasn't been touched in over 10 years ;-)

neheb, 2020-01-22:
:). I do agree that bsdtar is superior.

hnyman, 2020-01-22:
I think that the host build could be removed from the target package.
It and been added in 2017 to provide zip for others.
Not needed for the package itself.
And now there is the host tool.

Host addition in 2017
https://github.com/openwrt/packages/commit/7313348033e1bc53f0d4346036b5f98562b9379d#diff-d4d5e2fde1a43c1e588625e656ddc9f8

neheb, 2020-01-22:
https://github.com/openwrt/packages/pull/11089

It's still not being built. I have no idea why.

hnyman, 2020-01-23:
You should separate the luarocks commit from the PR and commit luarocks separately (as that change corresponds to the already existing zip host tool froim tools).

Then PR 11089 would be just about zip. Right the lua things pollute and expand the build logs quite much.

neheb, 2020-01-27:
This is still a problem btw.

4k3or3et, 2020-01-31:
On OpenWrt 19.07.1 the problem is exactly the same. Any other fixes can be tried to resolve the issue?

neheb, 2020-01-31:
I can backport my commit that had some bugfixes. But the problem would be the same. I have no idea how to fix.

I think bsdtar is the best solution here.

hnyman, 2020-02-01:
I wonder if we should just rename this package feed zip to "infozip".
That should remove the name collision.

BKPepe, 2020-02-22:
@hnyman can we close this?

closed 2020-02-23 by hnyman