openwrt/packages #10987: Libopenssl 1.1.1 dependence absence

y-alexey, 2020-01-11:
For mipsel24kc libopenssl required kmod-cryptodev. But this kmod is not in repo.

neheb, 2020-01-11:
ping @cotequeiroz

neheb, 2020-01-14:
Note that as an alternative, you can use AFALG.

In master, there is also afalg_engine , which is faster than the OpenSSL included version.

cotequeiroz, 2020-01-15:
It seems that cryptodev-linux is being built, but the binary package does not show up in the download directory.  When building eventually fails, a faillog gets generated, so the package not getting compiled does not appear to be the case.  Also, openssl would not even build if `package/cryptodev-linux/compile` should fail.  Also, the `cryptodev.h` file needs to be installed for the libopenss-devcrypto compilation to succeed.
This has been happening for a while.  I've looked for a kmod-devcrypto in a number of targets (mvebu, for example), and the package is missing in all repositories from lede-17.01 to master.  I guess people never realized this because cryptodev-linux was not listed as a dependency to openssl before 19.07.
Is this behavior intended, as if cryptodev-linux were supposed to be source-only?  If so, we need to undo it.

y-alexey, 2020-01-15:
Thank you for your valuable comment.  However, I did not understand how the situation you described affects the functioning of hardware acceleration of cryptography in openwrt.
 If the actual absence of the module did not affect, and now does not affect the acceleration of cryptography, then I will simply ignore the fact of detecting a dependency error until the miners figure it out. 
If as a result of this incident in the latest release there is actually no hardware acceleration, this is a completely different story.  And, at a minimum, it is necessary to warn people about this.

cotequeiroz, 2020-01-15:
@y-alexey 
Sorry, most of my comment was directed more towards the developers than the users.  It is a bug that prevents userspace use of crypto hardware through /dev/crypto, and it should be fixed.

Now here's some useful info for you in particular.

HW crypto acceleration is done by the kernel.  These drivers are built into the kernel on boards that actually have hw-crypto.  So kernel-mode acceleration works by default.  An example of application that uses this is ipsec ESP.  
You may check the available kernel ciphers in `/proc/crypto`.  There are multiple drivers implementing the same algorithms.  They are sorted by `priority`.  Accelerated drivers usually have a priority >= 300.  I am explaining this because apparently there is no hw-crypto support for anything with a mips CPU.  Granted, I'm not the best person to know what's in all supported boards, but I got that by running this:
```
git grep '^CONFIG_CRYPTO_\(HW\|DEV\)' -- $(git grep 'ARCH.*mips' | sed 's#^\([^/]\+\).*#\1#' | uniq)
```
So make sure you actually have hw-crypto first.
What `kmod-cryptodev` does is expose this functionality to user-space applications, by creating a `/dev/crypto` device.  Note that the mere presence of the `/dev/crypto` device does not mean that you have hw-crypto.  Cryptodev exports soft and hw-accelerated ciphers.  Applications could use `/dev/crypto` directly, but they usually use the crypto functions provided by a crypto library, such as openssl.

Then the openssl devcrypto engine (`libopenssl-devcrypto` package) uses /dev/crypto to make any application using openssl perform the crypto operations in hw.
`/dev/crypto` is not the only way that openssl can use hw-crypto.  As @neheb pointed out, there's the AF_ALG interface, which is the "official" linux userspace crypto implementation.  It performs the same basic functions of `/dev/crypto`, but using sockets instead of a device, so you'll see nothing under `/dev`.  `AF_ALG` is provided by the `kmod-crypto-user` package, which does not have the bug you've found.  To use it, install `libopenssl-afalg` instead of `libopenssl-devcrypto`.  It will use the kernel to perform AES-CBC only, and will not detect if it is hardware-accelerated or not.  It is slower than /dev/crypto as well.

@neheb suggested to use an alternative afalg engine, but it is not available in 19.07.  It is basically a mirror of the devcrypto engine using the AF_ALG interface, allowing you to use only hw-accellerated drivers.  You may install it in 19.07 straight from the snapshots repo with:
`opkg update; opkg install 'http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/libopenssl-afalg_sync_1.0.1-1_mipsel_24kc.ipk'`).

There's more information [in openwrt documentation](https://openwrt.org/docs/techref/hardware/cryptographic.hardware.accelerators).  It does not show the new afalg engine yet (I'll edit it and include at least something basic).  Basically, to configure the afalg engine, substitute `devcrypto` with `afalg` in `/etc/ssl/openssl.cnf`, when following the instructions.  Note that the drivers name will not be shown because the `kmod-crypto-user` package in 19.07, ironically, does not include the `crypto_user.ko` module.

cotequeiroz, 2020-01-15:
@Ansuel
jow explained to me on IRC that the redefinition of `DEFAULT:=m if ALL` "overwrites the default `DEFAULT` selection logic, so it fails to build in the phase1 builders".  I imagine that its intent was to not build the package by the buildbots, so it could only be compiled locally.  It made sense before the engines were put in a separate package--you'd have to compile openssl yourself to enable it, so you might as well compile devcrypto along--but it does not anymore. I'm testing the package without that line, and send a patch to the ML if OK.  What are your thoughts as the maintainer?

Ansuel, 2020-01-15:
@cotequeiroz LGTM

cotequeiroz, 2020-01-15:
The patch to fix this has been accepted.  I'm waiting to see the package in 'snapshots' before closing this.

cotequeiroz, 2020-01-15:
The kmod-cryptodev packages are starting to show up in the snapshot builds.  This has been fixed in master, and applied to 19.07 as well, where it should first appear in 19.07.1.

closed 2020-01-15 by cotequeiroz

jow-, 2020-01-16:
The `kmod-cryptodev` package is getting built now, e.g. https://downloads.openwrt.org/snapshots/targets/ath25/generic/packages/kmod-cryptodev_4.14.162+1.10-ath25-1_mips_mips32.ipk so this issue can likely be closed.