openwrt/packages #10992: vpn-policy-routing recursive dependency on 1907

manjuprajna, 2020-01-12:
Maintainer: @stangri 

Environment: (put here arch, model, OpenWrt version)
build OpenWrt 1907 on Ubuntu 1804

Description:
when run make menuconfig see below error:
tmp/.config-package.in:135802:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:135802:  symbol PACKAGE_vpn-policy-routing depends on PACKAGE_vpnbypass
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:135822:  symbol PACKAGE_vpnbypass depends on PACKAGE_vpn-policy-routing

stangri, 2020-01-12:
@jow- is there a better solution than to uninstall one of the packages thru ```./scripts/feeds remove```?

hnyman, 2020-01-12:
The strange part is that the error is caused by the mirrored CONFLICTS statements, right?
I guess that removing one of those would be enough.

neheb, 2020-01-17:
https://github.com/openwrt/packages/commit/3030d0fc1af6e21deab3b16304c2e524f1f724a7

closed 2020-01-17 by neheb