openwrt/packages #10993: gpsd: build fails as of missing python distutils.setuptools

SvenRoederer, 2020-01-12:
Maintainer: @psidhu @p-wassi 
Environment: OpenWrt master (around 1.1.2020); mvebu-cortexa9

Description: building gpsd fails with a missing dependency to python module distutils.sysconfig

```
scons: Reading SConscript files ...
ImportError: cannot import name 'sysconfig' from 'distutils' (/usr/lib/python3.7/distutils/__init__.py):
  File "/mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/gpsd-3.19/SConstruct", line 46:
    from distutils import sysconfig
make[3]: *** [Makefile:148: /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/gpsd-3.19/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[3]: Leaving directory '/mnt/local/btrfs/build/freifunk/ffberlin/firmware/packages/packages/utils/gpsd'
time: package/feeds/packages/gpsd/compile#0.42#0.10#0.51
make[2]: *** [package/Makefile:113: package/feeds/packages/gpsd/compile] Error 2
```

Installing the python-module via the hosts apt-get fixes the build, but this seems not the way to go.

neheb, 2020-01-14:
ping @commodo , @RussellSenior

jefferyto, 2020-01-14:
The error came from the SConstruct file, which means it was run by scons.

AFAIK the [scons package](https://github.com/openwrt/packages/blob/3aa52515720b75ada15777ab8d14c1d253921638/devel/scons/Makefile) depends on buildhost Python, not host Python provided by the python package.

Your buildhost, it has distutils installed (in /usr/lib/python3.7/distutils) but not the sysconfig.py file (/usr/lib/python3.7/distutils/sysconfig.py)? What distro / OS version are you using?

SvenRoederer, 2020-01-14:
I'm running Debian 10.2 with small packageset installed, so "python3-distutils" wasn't installed at all.

jefferyto, 2020-01-14:
I've set up a minimal Debian 10.2 VM and it has python3.7-minimal, which only has \_\_init__.py and version.py inside /usr/lib/python3.7/distutils.

I've tried compiling the gpsd package (using a snapshot sdk), but before it can get to it, the host scons package throws an error:

```
./files/pywrap.sh /home/jeff/ath79/build_dir/hostpkg/scons-3.1.1/setup.py install --prefix=/home/jeff/ath79/staging_dir/host
Traceback (most recent call last):
  File "/home/jeff/ath79/build_dir/hostpkg/scons-3.1.1/setup.py", line 26, in <module>
    import distutils.command.build_scripts
ModuleNotFoundError: No module named 'distutils.command'
```

How did you manage to install host scons without full distutils?

SvenRoederer, 2020-01-15:
Probably scons was built once I had "python3-distutils" installed when tracing down the build-error. Not sure if I build all from scratch then ...

SvenRoederer, 2020-01-16:
@jefferyto You are right. I started the build from scratch and have now scons failing to build.

```
make[3]: Entering directory '/mnt/local/btrfs/build/freifunk/ffberlin/firmware/packages/packages/devel/scons'
./files/pywrap.sh /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/hostpkg/scons-3.1.1/setup.py install --prefix=/mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/staging_dir/host
Traceback (most recent call last):
  File "/mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/hostpkg/scons-3.1.1/setup.py", line 26, in <module>
    import distutils.command.build_scripts
ModuleNotFoundError: No module named 'distutils.command'
make[3]: *** [Makefile:56: /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/staging_dir/hostpkg/stamp/.scons_installed] Error 1
make[3]: Leaving directory '/mnt/local/btrfs/build/freifunk/ffberlin/firmware/packages/packages/devel/scons'
time: package/feeds/packages/scons/host-compile#0.25#0.13#0.36
make[2]: *** [package/Makefile:111: package/feeds/packages/scons/host/compile] Error 2
```

closed 2020-01-20 by neheb

SvenRoederer, 2020-01-25:
@jefferyto  thanks for taking care.

PR #11040 (2020-01-20): gpsd: Update to 3.20, add/refresh patches
Maintainer: @psidhu 
Compile tested: armvirt-64, 2020-01-15 snapshot sdk
Run tested: none

Description:
New patches (both for SConstruct):

* Do not import from distutils. Since Python support/modules are not built, there is no need to import from distutils. (Importing from distutils may prevent the package from being built on systems without a full distutils module, e.g. Debian with python3-minimal.)

  I have added back the import in places where it may be helpful to have, if scons is run manually.

  Fixes #10993.

* Do not check the size of `time_t [... cut]
commit a29905c4 (2020-01-20): gpsd: Update to 3.20, add/refresh patches

New patches (both for SConstruct):

* Do not import from distutils. Since Python support/modules are not
  built, there is no need to import from distutils. (Importing from
  distutils may prevent the package from being built on systems without
  a full distutils module, e.g. Debian with python3-minimal.)

  I have added back the import in places where it may be helpful to
  have, if scons is run manually.

  Fixes #10993.

* Do not check the size of time_t. The way this is done is to compile
  and run a test program; this fails when cross-compiling[1 [... cut]