openwrt/packages #10997: adblock shell error

ifuchs, 2020-01-12:
I am attempting to run adblock on an old NSLU2 with OpenWRT. The installs run fine and I can start the service bur status shows:

root@OpenWrt:~# /etc/init.d/adblock status
::: adblock runtime information
  + adblock_status  : running
  + adblock_version : 3.5.5-2
  + overall_domains : 0 (normal mode)
  + fetch_utility   : /bin/uclient-fetch (-)
  + dns_backend     : dnsmasq (/tmp)
  + last_rundate    : 12.01.2020 12:44:25
  + system_release  : Linksys NSLU2, OpenWrt 18.06.5 r7897-9d401013fc

Also, the log has:
root@OpenWrt:~# logread -f -e"adblock"
Sun Jan 12 12:55:29 2020 daemon.err adblock.sh[1441]: sh: 1:wq: bad number
Sun Jan 12 12:55:30 2020 user.info adblock-3.5.5-2[1441]: adblock instance started ::: action: start, priority: 0, pid: 1441
Sun Jan 12 12:55:30 2020 daemon.err adblock.sh[1441]: sh: 1:wq: bad number
Sun Jan 12 12:55:30 2020 daemon.err adblock.sh[1441]: sh: 1:wq: bad number
Sun Jan 12 12:55:30 2020 daemon.err adblock.sh[1441]: sh: 1:wq: bad number
Sun Jan 12 12:55:41 2020 user.info adblock-3.5.5-2[1441]: blocklist with overall 0 domains loaded successfully (Linksys NSLU2, OpenWrt 18.06.5 r7897-9d401013fc)

What have I overlooked?

dibdot, 2020-01-12:
Please provide your adblock config (global and extra section only) - thanks.

ifuchs, 2020-01-12:
config adblock 'global'
	option adb_enabled '1:wq'
	option adb_dns 'dnsmasq'
	option adb_fetchutil 'uclient-fetch'
	option adb_trigger 'wan'

config adblock 'extra'
	option adb_debug '0'
	option adb_forcesrt '0'
	option adb_forcedns '0'
	option adb_backup '0'
	option adb_maxqueue '4'

dibdot, 2020-01-12:
`option adb_enabled '1:wq'`
looks like an editing typo, this option expects a bool value, e.g.
`option adb_enabled '1'`
is valid and should work.
Please note: your config lacks ssl support, please install 'libustream-mbedtls' to support ssl/https downloads.

closed 2020-01-12 by dibdot

ifuchs, 2020-01-12:
My bad (:eq was meant to go to vi). Now that the error is resolved, I find that the adblocker is not blocking any ads. Perhaps there is a problem getting the lists? How do I go about debugging this?
Also when I tried to install luci-app-adblock, it installs wihtout wrror but then the log begins to show this error repeatedly:

Sun Jan 12 18:04:24 2020 daemon.err uhttpd[1933]: The file '/usr/lib/lua/luci/controller/adblock.lua' contains an invalid module line.
Sun Jan 12 18:04:24 2020 daemon.err uhttpd[1933]: Please verify whether the module name is set to 'luci.controller.adblock' - It must correspond to the file path!

Not sure how to fix that (to which file path is it referring).
So bottom line so far is adblock and the luci addon are installed but ther eis no filetering and the web interface is not working.

ifuchs, 2020-01-13:
Further: I realized that adblock was enabled but not started! I started it and the log now says:

root@OpenWrt:~# logread -f -e "adblock"
Sun Jan 12 19:06:50 2020 user.info adblock-3.5.5-2[13482]: adblock instance started ::: action: start, priority: 0, pid: 13482
Sun Jan 12 19:06:57 2020 user.info adblock-3.5.5-2[13482]: blocklist with overall 0 domains loaded successfully (Linksys NSLU2, OpenWrt 18.06.5 r7897-9d401013fc)

As I suspected, it is not loading the blocklists.

ifuchs, 2020-01-13:
Added libustream-mbedtls. Now I have blocklists loading:
root@OpenWrt:~# /etc/init.d/adblock status
::: adblock runtime information
  + adblock_status  : enabled
  + adblock_version : 3.5.5-2
  + overall_domains : 30210 (normal mode)
  + fetch_utility   : /bin/uclient-fetch (libustream-ssl)
  + dns_backend     : dnsmasq (/tmp)
  + last_rundate    : 12.01.2020 19:15:33
  + system_release  : Linksys NSLU2, OpenWrt 18.06.5 r7897-9d401013fc
Tested with: https://ads-blocker.com/testing/ and seems to be working.
Haven't figured out the luci-app-adblock error.

Thank you @dibdot