openwrt/packages #11007: Add an helloworld package

zoobab, 2020-01-13:
It would be great to add an helloworld package, maybe in C to start with.

Inspired from the documentation:

https://openwrt.org/docs/guide-developer/helloworld/chapter2

karlp, 2020-01-13:
I feel the purpose of that guide is to show you how to do exactly that.  Including it directly would just cause conflicts?  Perhaps you can outline more how this would help?  If you're learning to develop the packaging, you don't want it already existing, and if you aren't, then there's .... not much use in being able to opkg install helloworld that I can see?

zoobab, 2020-01-13:
Well that validates that you can compile and run helloworld in C.

I found the example of using the SDK very very cumbersome, and the helloworld package is the simplest example I can find. Users of the SDK recommend to compile the `firewall` package, which has lots of dependencies to the kernel.

While the package is mostly useless in itself, it allows you to validate your tools, toolchain, and that the execution on your target processor works.

jefferyto, 2020-01-27:
So what you really want is a way to know your buildhost is configured correctly, not a beginner example of how to create a package?

zoobab, 2020-01-27:
Both, for example copy/pasting the example ended up, after several minutes of debugging to know where was the error, in a "yes, but copy/pasting the example does not include TABS" :-).

If that was present in GIT, the error would not happen.

It is especially confusing for a newbie.

Furthermore, if you try the helloworld example with the SDK, it will fail because of missing dependencies towards the libc package.

I would even go further, and name the package as "helloworld-sh", so that there are no dependencies, with a simple helloworld in shell.

And then make another package for helloworld in all other supported languages, like:

* helloworld-c
* helloworld-sh
* helloworld-go
* helloworld-lua

etc...

jefferyto, 2020-01-27:
Even if you had example code, you would have had to learn about tabs in Makefiles at some point. Having example code that works only delays this learning until later.

This sounds like an opportunity for you to improve the documentation on the wiki, e.g. the [Creating packages](https://openwrt.org/docs/guide-developer/packages) page.

On making sure your system has the necessary build requirements, this sounds like the [Build system - Installation](https://openwrt.org/docs/guide-developer/build-system/install-buildsystem) page can be improved. In fact it sounds like the [Hello world](https://openwrt.org/docs/guide-developer/helloworld/start) guide on the wiki can be improved as well.

Maybe example code can help, but I think its place would be in the wiki or in a separate example feed repo, not here. The wiki is for documentation, the packages repo is for shipping packages.

diizzyy, 2020-01-29:
This is a no go as long as there's no documentation that explains the structure of a Makefile, variables and their proper values and so on.

BKPepe, 2020-02-22:
I agree with @jefferyto. Closing.

closed 2020-02-22 by BKPepe