openwrt/packages #11008: libffi missing includes

erdoukki, 2020-01-13:
Using OpenWRT 19.07.0 on MVEBU...
libffi missing includes files and make python3 compilations may broken...
Also installed python3-cffi-src

neheb, 2020-01-14:
ping @commodo

commodo, 2020-01-14:
will take a look;
i'm assuming this happens during build-time;

did the buildbots catch anything?

erdoukki, 2020-01-14:
May be I have made a too much quick explanation...
The libffi package is installing correctly, what is missing is compiling something needing libffi headers...
python-dev build for sample, is bugging when using pip to install cffi...
I can give a try to show you some log if needed...

I do not find any package containing libffi headers so I open this issue...

commodo, 2020-01-14:
so, is this issue on the target? or on the host;

if you need python3-cffi on the target you can install it via  `opkg install python3-cffi`
installing it via pip on the target is a bit complicated;

erdoukki, 2020-01-15:
yes the issue is on target...
pip sometimes need specific version, like my try with mitmproxy...
I already `opkg install python3-cffi` and `opkg install python3-cffi-src` and `opkg install libffi` but still searchin a package which install ffi.h so I add one manually from my debian computer...
now the missing is libffi itself, I get errors from the `pip install cryptograpgy` ;
`/usr/bin/ld: cannot find -lffi`

erdoukki, 2020-01-15:
I also add the missing `/usr/lib/pkgconfig/libffi.pc` from my debian PC
`# cat /usr/lib/pkgconfig/libffi.pc`
```
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
toolexeclibdir=${exec_prefix}/lib/../lib
includedir=${prefix}/include

Name: libffi
Description: Library supporting Foreign Function Interfaces
Version: 3.2.1
Libs: -lffi
Cflags:

```

erdoukki, 2020-01-15:
```
...
/usr/bin/ld: cannot find -lffi
collect2: error: ld returned 1 exit status
error: command 'aarch64-openwrt-linux-musl-gcc' failed with exit status 1
```

```
root@monitor:~/cffi-1.13.2# opkg files libffi
Package libffi (3.2.1-3) is installed on root and has the following files:
/usr/lib/libffi.so.6
/usr/lib/libffi.so.6.0.4

```

jow-, 2020-01-15:
Compilation on-target is not really supported by OpenWrt.

jefferyto, 2020-01-27:
> pip sometimes need specific version, like my try with mitmproxy...

What you really want is to install mitmproxy, is this correct?

IIRC from investigating #8151, mitmproxy's dependencies are a bit conservative... In particular, the current version (5.0.1) wants [`cryptography>=2.1.4,<2.5`](https://github.com/mitmproxy/mitmproxy/blob/v5.0.1/setup.py#L68) whereas the current version of cryptography is 2.8. (So using pip to install mitmproxy, pip will try to install cryptography 2.4.2 even if you have our python-cryptography package installed because the package is too new, which causes pip to try to build libffi on device.)

erdoukki, 2020-01-27:
> What you really want is to install mitmproxy, is this correct?
Yes...
But so bad that python-pip hang just for some headers missing files ...?

jefferyto, 2020-01-27:
Without a full compile/install log, it would be impossible to know what is causing your problem. But as @jow- mentioned, compilation on-target is not really supported.

If you want to install mitmproxy, you should create a package (i.e. Makefile, patches, etc.), e.g. [radicale2](https://github.com/openwrt/packages/tree/master/net/radicale2),  then (cross-)compile the package [using the SDK](https://openwrt.org/docs/guide-developer/using_the_sdk).

neheb, 2020-02-10:
Closing as this is not really supported.

If you want mitmproxy, propose a package for it as a PR.

closed 2020-02-10 by neheb