openwrt/packages #11014: Tor Lack of multithread support

y-alexey, 2020-01-14:
For mipsel_kc24 architecture package tor (oblivious) make with one thread available. Or, may be, need some unpublished options to start multithread. That's lead to 
[image]100% load of one core.
Is this can be solved anyway?

neheb, 2020-01-15:
Does tor even support multithreading? I would think no.

y-alexey, 2020-01-15:
> Does tor even support multithreading?
First Google link "tor multithread" is to tor bug tracker about this problem 10 year ago. And it was solved, as I can see.

jefferyto, 2020-01-27:
I believe this is the ticket you are referring to: https://trac.torproject.org/projects/tor/ticket/1749

If the problem was "solved" then the ticket would have been marked "resolved".

It doesn't mention Ubuntu (or Debian) at all. How do you know it is multithreaded on Ubuntu?

y-alexey, 2020-01-27:
Looked like you right. Work on the go. https://trac.torproject.org/projects/tor/ticket/26296

closed 2020-01-27 by y-alexey