openwrt/packages #11022: miniupnpd: external_iface is not wan as default

ogarcia, 2020-01-14:
miniupnpd version 2.1.20190408-2
OpenWrt 19.07.0 on AC1750

In my setup I have two interfaces, wan attached to eth0.6 that provides Internet connection and voip attached to eth0.3 that only provides an internal address to connect to my carrier SIP server. If you read [documentation about miniupnpd][1] it says:
external_iface, type string, no required, default wan

This is not true because if I not define `external_iface` then it picks voip/eth0.3 instead of wan/eth0.6, but if I add `option external_iface 'wan'` in config file it picks the correct interface and works well.

[1]: https://openwrt.org/docs/guide-user/firewall/upnp/miniupnpd

FunnyAnimal, 2020-01-31:
I had the same issue. miniupnpd does "smart" guessing of what it thinks is the wan side of your system.

Your eth0.3 is probably misconfigured, and is causing a "default route" entry in the routing table. In my case, I had setup an internal interface with a gateway outside the subnet of the interface, like this:

```
cat /etc/config/network
[...]
config interface 'lan3'
        option ifname 'br-lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option ipaddr '192.168.3.1'
-->     option gateway '192.168.0.1'  <--
```

This would cause an extra default route via 'lan3' and miniupnpd to choose 'lan3', thus 'br-lan', as its wan-side.

The solution in my case was to remove the (incorrect) 'option gateway' setting for 'lan3'.

ogarcia, 2020-01-31:
In my case the interface have a gateway (configured via dhcp):
```
config interface 'voip'
	option ifname 'eth0.3'
	option proto 'dhcp'
```
It has not default gateway in `ip ro` command, but you have me an idea. I see the interface via `ifstatus` and _there is_ a default route:
```
{
	"target": "0.0.0.0",
	"mask": 0,
	"nexthop": "10.27.160.1",
	"source": "10.27.173.208/32"
}
```
I add `option defaultroute '0'` to interface config
```
config interface 'voip'
	option ifname 'eth0.3'
	option proto 'dhcp'
	option defaultroute '0'
```
And now can remove `option external_iface 'wan'`.

Anyway I think that the issue continues becasue if _default_ behaviour is that external interface is `wan` so would not be neccesary set option explicit. Or change documentation and write, as default it pick the first interface with default route.

FunnyAnimal, 2020-01-31:
Yes, i agree: "default wan" as stated in documentation is in fact incorrect. The default is "the first interface that offers a default route". (Or maybe even "a randomly selected interface that offers a default route"?) Which is usually 'wan' but not always, depending on network configuration. Might want to ping @jow- as maintainer of miniupnpd.

jow-, 2020-01-31:
Which documentation do you refer to?

ogarcia, 2020-01-31:
@jow- I put in my first comment :-)
https://openwrt.org/docs/guide-user/firewall/upnp/miniupnpd -> external_iface, type string, no required, **default wan**

FunnyAnimal, 2020-01-31:
As mentioned above:

> In my setup I have two interfaces, wan attached to eth0.6 that provides Internet connection and voip attached to eth0.3 that only provides an internal address to connect to my carrier SIP server. If you read [documentation about miniupnpd](https://openwrt.org/docs/guide-user/firewall/upnp/miniupnpd) it says:
> external_iface, type string, no required, default wan
> 

external_iface | string | no | wan | External interface (wan)

jow-, 2020-01-31:
Documentation fixed.

closed 2020-01-31 by jow-

FunnyAnimal, 2020-01-31:
Wow! That is fast! My reloading of that page took longer than your fix :-D