openwrt/packages #11034: fwknop - /etc/init.d/fwknopd: line 11: /lib/functions/network.sh: No such file or directory

PeteLawler, 2020-01-16:
Maintainer: @jp-bennett
Environment: Building 19.07.0 for ath79 (though not architecture specific issue)

Description:
Error thrown during build.
```
openwrt-imagebuilder-19.07.0-ath79-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ath79/etc/init.d/fwknopd: line 11: /lib/functions/network.sh: No such file or directory
./etc/init.d/fwknopd: line 11: /lib/functions/network.sh: No such file or directory
Formating code blocks by wrapping them with pairs of ```
```

I'm not overly familiar with the repos and how stuff is built, but a glance makes me think that it is possible that an attempted fix was made in 5a69b9c08f9bc9d5b8fd868648b418075a1057aa, however was rejected by Travis https://travis-ci.org/openwrt/packages/builds/288644851

jp-bennett, 2020-01-16:
Yes, and Travis failed it for a totally unrelated build error. Thanks for reporting, I'll get this fixed.

I'm not overly familiar with the repos and how stuff is built, but a glance makes me think that it is possible that an attempted fix was made in 5a69b9c, however was rejected by Travis https://travis-ci.org/openwrt/packages/builds/288644851

PeteLawler, 2020-01-16:
Cheers, thanks for that. Would you like me to open a separate issue to get the package version bumped from 2.6.9 to match upstream (https://github.com/mrash/fwknop is 2.6.11) or you happy to take my mention of it in passing? ;-)

jp-bennett, 2020-01-16:
I also see where I did a pull request to bump it to 2.6.10, which also never got pulled. I got extremely busy with another project, and didn't follow up. Mentioning it in passing is sufficient, thanks!

lkraav, 2020-05-26:
Hi all. Did anything end up happening here? I just built latest 19.07.3 image and issue subject error is still there :thinking: