openwrt/packages #11042: <ddns-scripts> ddns-scripts_cloudflare.com-v4 can not updating wildcard dns

keitetran, 2020-01-17:
DNS provider: Cloudflare
Device: Netgear Nighthawk X4S R7800
Firmware: LUCi
Version: OpenWrt 19.07.0 r10860-a3ffeb413b / LuCI openwrt-19.07 branch git-20.010.60094-335692a
Packages installed:
```
luci-app-ddns	2.4.9-7
ddns-scripts 2.7.8-12
ddns-scripts_cloudflare.com-v4	2.7.8-12
```

Cloudflare record config: 
```
A ------ * ------ 123.456.789
```

Luci DDNS setup:
```
DDNS Service provider [IPv4]: Cloudflare
Domain: *@mydomain.com
Username: My email
Password: Global API key
```

i can not start Process ID and get error:
```
135229       : ************ ************** ************** **************
135229  note : PID '11167' started at 2020-01-17 13:52
135229       : ddns version  : 2.7.8-12
135229       : uci configuration:
ddns.cloudflase.domain='*@asssssss.xyz'
ddns.cloudflase.enabled='1'
ddns.cloudflase.lookup_host='ip.asssssss.xyz'
ddns.cloudflase.password='***PW***'
ddns.cloudflase.service_name='cloudflare.com-v4'
ddns.cloudflase.username='asssssss@gmail.com'
ddns.cloudflase=service
135229       : verbose mode  : 0 - run normal, NO console output
135229  CRIT : sanitize on domain found characters outside allowed subset - TERMINATE
135229  WARN : PID '11167' exit WITH ERROR '1' at 2020-01-17 13:52
```

Try change domain name to: *.mydomain.com (Removed @)
I get same error message

Try change domain name to: mydomain.xyz
I get new error message
```
 135812       : ************ ************** ************** **************
 135812  note : PID '11966' started at 2020-01-17 13:58
 135812       : ddns version  : 2.7.8-12
 135812       : uci configuration:
ddns.cloudflase.domain='mydomain.xyz'
ddns.cloudflase.enabled='1'
ddns.cloudflase.lookup_host='ip.mydomain.xyz'
ddns.cloudflase.password='***PW***'
ddns.cloudflase.service_name='cloudflare.com-v4'
ddns.cloudflase.username='mydomain@gmail.com'
ddns.cloudflase=service
 135812       : verbose mode  : 0 - run normal, NO console output
 135812       : check interval: 600 seconds
 135812       : force interval: 259200 seconds
 135812       : retry interval: 60 seconds
 135812       : retry counter : 0 times
 135812       : No old process
 135812       : last update: never
 135812       : Detect registered/public IP
 135812       : #> /usr/bin/nslookup ip.mydomain.xyz  >/var/run/ddns/cloudflase.dat 2>/var/run/ddns/cloudflase.err
 135812       : Registered IP '133.208.456.132' detected
 135812  info : Starting main loop at 2020-01-17 13:58
 135812       : Detect local IP on 'network'
 135812       : Local IP '133.208.456.132' detected on network 'wan'
 135812       : Forced Update - L: '133.208.456.132' == R: '133.208.456.132'
 135813       : parsing script '/usr/lib/ddns/update_cloudflare_com_v4.sh'
 135813       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflase.dat --stderr /var/run/ddns/cloudflase.err --noproxy '*' --header 'X-Auth-Email: mydomain@gmail.com'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=mydomain.xyz'
 135813       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflase.dat --stderr /var/run/ddns/cloudflase.err --noproxy '*' --header 'X-Auth-Email: mydomain@gmail.com'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones/sss/dns_records?name=mydomain.xyz&type=A'
 135813  WARN : Could not detect 'record id' for host.domain.tld: 'mydomain.xyz'
 135814 ERROR : No update send to DDNS Provider
 135814       : Waiting 600 seconds (Check Interval)
```

keitetran, 2020-01-17:
```
# cloudflase.dat file content
{"result":[],"result_info":{"page":1,"per_page":20,"total_pages":0,"count":0,"total_count":0},"success":true,"errors":[],"messages":[]}
```
Script can not get recordID with * character.

keitetran, 2020-01-17:
```
__RUNPROG="$__PRGBASE --request GET '$__URLBASE/zones/$__ZONEID/dns_records?name=$__HOST&type=$__TYPE'"
```
change to
```
__RUNPROG="$__PRGBASE --request GET '$__URLBASE/zones/$__ZONEID/dns_records?name=*.$__HOST&type=$__TYPE'"
```
Working fine
```
 142324       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflase.dat --stderr /var/run/ddns/cloudflase.err --noproxy '*' --header 'X-Auth-Email: asddsfsdf@gmail.com'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones/asddsfsdf/dns_records?name=*.asddsfsdf.xyz&type=A'
 142324       : IPv4 at CloudFlare.com already up to date
 142325  info : Forced update successful - IP: '123.456.789' send
 142325       : Waiting 600 seconds (Check Interval)
```

closed 2020-01-28 by neheb

Ansuel, 2020-01-28:
Notice now that this broke my ddns configuration with cloudflare...

neheb, 2020-01-29:
:( should this be reverted?

keitetran, 2020-01-29:
Why you mege this commit ? 
It work only with user use wildcard dns. look like *.abc.com
Other user use A or CNAME record DNS: subdomain.abc.com
We need add a flag for wildcard dns or other. 
```
IF is wildcard dns 
   $__HOST = '*.' + $__HOST
END

__RUNPROG="$__PRGBASE --request GET '$__URLBASE/zones/$__ZONEID/dns_records?name=$__HOST&type=$__TYPE'"
```
OR remove domain validate at luci app via control panel. because it not accept charator * at domain field: *.abc.com is not valid.

Ansuel, 2020-01-29:
@keitetran i think the best way will be fix this in the ddns luci app since you could just manually change the ddns config file...
Can you test if with the old script and by manually edit the ddns config file all works correctly ?

keitetran, 2020-01-30:
im using manualy edit now and it working fine.

```
__RUNPROG="$__PRGBASE --request GET '$__URLBASE/zones/$__ZONEID/dns_records?name=*.$__HOST&type=$__TYPE'"
```

We need edit luci app validator to allow * character
With this config it working fine withh all DNS record config type 
`Domain: *@mydomain.com` or `Domain: subdomain@mydomain.com`
```
DDNS Service provider [IPv4]: Cloudflare
Domain: *@mydomain.com
Username: My email
Password: Global API key
```

Ansuel, 2020-01-30:
@neheb can you revert this? i will put a patch to the ddns luci app

Ansuel, 2020-01-30:
Also now that i check... in master this was already fixed...

neheb, 2020-01-31:
https://github.com/openwrt/packages/commit/81d67a21b136a58cd809263ad10db439d31b8e9a

Done. Sorry for the inconvenience.