openwrt/packages #11052: libtirpc: fail to build on musl-libc due to missing header

wongsyrone, 2020-01-19:
Maintainer: @Andy2244
Environment: x86_64

Description:

@Andy2244 @neheb 
Fail to build with musl-libc

```
OpenWrt-libtool: compile:  x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I../tirpc -include config.h -DPORTMAP -DINET6 -D_GNU_SOURCE -Wall -pipe -I/home/wong/github/lede-1/staging_dir/target-x86_64_musl/usr/include -I/home/wong/github/lede-1/staging_dir/toolchain-x86_64_gcc-9.2.0_musl/usr/include -I/home/wong/github/lede-1/staging_dir/toolchain-x86_64_gcc-9.2.0_musl/include/fortify -I/home/wong/github/lede-1/staging_dir/toolchain-x86_64_gcc-9.2.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/wong/github/lede-1/build_dir/target-x86_64_musl/libtirpc-1.2.5=libtirpc-1.2.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DGQ -MT libtirpc_la-xdr_float.lo -MD -MP -MF .deps/libtirpc_la-xdr_float.Tpo -c xdr_float.c  -fPIC -DPIC -o .libs/libtirpc_la-xdr_float.o
xdr_float.c:86:10: fatal error: bits/endian.h: No such file or directory
   86 | #include <bits/endian.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[6]: *** [Makefile:962: libtirpc_la-xdr_float.lo] Error 1
make[6]: Leaving directory '/home/wong/github/lede-1/build_dir/target-x86_64_musl/libtirpc-1.2.5/src'
make[5]: *** [Makefile:519: all-recursive] Error 1
make[5]: Leaving directory '/home/wong/github/lede-1/build_dir/target-x86_64_musl/libtirpc-1.2.5'
make[4]: *** [Makefile:404: all] Error 2
make[4]: Leaving directory '/home/wong/github/lede-1/build_dir/target-x86_64_musl/libtirpc-1.2.5'
make[3]: *** [Makefile:56: /home/wong/github/lede-1/build_dir/target-x86_64_musl/libtirpc-1.2.5/.built] Error 2
make[3]: Leaving directory '/home/wong/github/lede-1/feeds/packages/libs/libtirpc'
time: package/feeds/packages/libtirpc/compile#13.27#2.12#16.39
make[2]: *** [package/Makefile:113: package/feeds/packages/libtirpc/compile] Error 2
make[2]: Leaving directory '/home/wong/github/lede-1'
make[1]: *** [package/Makefile:107: /home/wong/github/lede-1/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wong/github/lede-1'
make: *** [/home/wong/github/lede-1/include/toplevel.mk:227: world] Error 2
```

The fix is simple, just change `#include <bits/endian.h>` to `#include <endian.h>`

neheb, 2020-01-20:
Is this on a newer version of musl?

wongsyrone, 2020-01-20:
I'm not sure, I'm using musl git commit: d01fdc777dd3b5ebcad351ee47d1984d28db31e4

wongsyrone, 2020-01-20:
I skimmed the commit log, it looks like the header is moved to alltypes.h via https://git.musl-libc.org/cgit/musl/commit/?id=97d35a552ec5b6ddf7923dd2f9a8eb973526acea

wongsyrone, 2020-01-20:
I think it should be considered as a bug and fixed upstream, glibc treats it as an error as well, you may check https://sourceware.org/git/gitweb.cgi?p=glibc.git;a=blob;f=bits/endian.h;h=45afd4ae47708c0c1953a06ed95aa6f3a8da302f;hb=HEAD

Andy2244, 2020-01-20:
@neheb ok [thanks](https://sourceforge.net/p/libtirpc/mailman/libtirpc-devel/?viewmonth=202001), if its not urgent i will just wait for the upstream fix?

neheb, 2020-01-23:
Fix merged.

closed 2020-01-23 by neheb