openwrt/packages #11055: Base System: USB has no power on TP-Link TL-WR842ND v2.0 after upgrading to 19.07

arminfuerst, 2020-01-19:
After upgrading from 18.06 to 19.07 the USB port of the device seems to provide no power at all. I have a simple device (TEMPer) I am using for this test. The device has a power LED which starts as soon as there is power on the USB port. The LED in on if I even only connect the device to an active USB hub not being connected to a computer.
I checked that the device has power just before the upgrade.
I made a factory reset of the device before the upgrade.
I tried with two identical devices and had the same effect.
When I run lsusb, I do see the USB hub, but no connected devices.
Can I provide any additional info to support working on this issue?

diizzyy, 2020-01-19:
Wrong repo, please use https://bugs.openwrt.org/

closed 2020-01-19 by diizzyy

arminfuerst, 2020-01-19:
Sorry and thank you. I just created it there:
https://bugs.openwrt.org/index.php?do=details&task_id=2753