openwrt/packages #11062: <radicale-py3> Missing dependency python3-urllib in 19.07.0

ddast, 2020-01-19:
Maintainer: @jefferyto
Environment: OpenWrt 19.07.0 (TP-Link C2600)

Description:
Updated to 19.07.0 and installed radicale-py3. Radicale does not start since the module urllib is missing.  This can be fixed by installing python3-urllib.

It looks like this was already fixed in [master](https://github.com/openwrt/packages/blob/master/net/radicale/Makefile) in commit 9c5a97c31974096d7d2a6f482dc78971b9b578d0, however, [branch 19.07 ](https://github.com/openwrt/packages/blob/openwrt-19.07/net/radicale/Makefile)does not have python3-urllib as a dependecy.  Actually the mentioned commit replaces the dependency python3-email with python3-light and python3-urllib.

jefferyto, 2020-01-19:
Thanks for the report - sounds like you know exactly what needs to be updated (replacing the python3-email dependency with python3-urllib should be enough). Do you want to open a PR with this change?

(As per the [Contributing Guidelines](https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md), you'll also need to increment `PKG_RELEASE` and include a `Signed-off-by` tag in the commit message.)

ddast, 2020-01-20:
Thanks for the fast reply.  I created PR #11078.

jefferyto, 2020-01-21:
With #11078 merged, this should be fixed.

PR #11078 (2020-01-20): radicale-py3: Add python3-urllib dependency
Depend on python3-urllib instead of python3-email (python3-urllib has
python3-email as a dependency).

Maintainer: @jefferyto
Compile tested: -
Run tested: TP-Link Archer C2600, ipq806x, OpenWrt 19.07.0. Confirmed that with the current dependencies radicale-py3 only starts after manually installing python3-urllib.

Description:
Radicale does not start with the current dependencies since urllib is missing.  After installing python3-urllib, radicale-py3 works.
This was already fixed in master in commit 9c5a97c, however, branch 19.07 does not yet have p [... cut]
closed 2020-01-21 by neheb