openwrt/packages #11067: QEMU not compiling on multiple platforms

neheb, 2020-01-20:
Maintainer: @yousong 
Environment: ath79, aarch64

https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/qemu/compile.txt

```
util/mmap-alloc.c: In function 'qemu_ram_mmap':
util/mmap-alloc.c:144:26: error: 'MAP_SYNC' undeclared (first use in this function); did you mean 'MS_SYNC'?
  144 |         map_sync_flags = MAP_SYNC | MAP_SHARED_VALIDATE;
      |                          ^~~~~~~~
      |                          MS_SYNC

```

I have fortify-headers disabled locally. Hence the error.

yousong, 2020-01-20:
```
install: cannot stat '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_generic_musl/qemu-4.2.0/ipkg-install/usr/bin/qemu-ga': No such file or directory
Makefile:435: recipe for target '/builder/shared-workdir/build/sdk/bin/packages/aarch64_generic/packages/qemu-ga_4.2.0-1_aarch64_generic.ipk' failed
make[3]: *** [/builder/shared-workdir/build/sdk/bin/packages/aarch64_generic/packages/qemu-ga_4.2.0-1_aarch64_generic.ipk] Error 1
time: package/feeds/packages/qemu/compile#55.74#12.73#77.95
```

neheb, 2020-03-29:
I still get this error.

yousong, 2020-03-30:
Please be specific.  I didn't see qemu error in the following links

 - https://downloads.openwrt.org/snapshots/faillogs/mips_24kc/packages/
 - https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/

neheb, 2020-03-30:
This is the log:

```
mips-openwrt-linux-musl-gcc -iquote /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/util -iquote util -iquote /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/tcg -iquote /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/tcg/mips -I/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/linux-headers -I/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/linux-headers -iquote . -iquote /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0 -iquote /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/accel/tcg -iquote /home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/include -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include  -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/include/glib-2.0 -pthread -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/lib/glib-2.0/include   -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -Wstrict-prototypes -Wredundant-decls -Wall -Wundef -Wwrite-strings -Wmissing-prototypes -fno-strict-aliasing -fno-common -fwrapv -std=gnu99   -Wexpansion-to-defined -Wendif-labels -Wno-shift-negative-value -Wno-missing-include-dirs -Wempty-body -Wnested-externs -Wformat-security -Wformat-y2k -Winit-self -Wignored-qualifiers -Wold-style-declaration -Wold-style-definition -Wtype-limits -I/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/tests -I qga/qapi-generated -MMD -MP -MT util/mmap-alloc.o -MF util/mmap-alloc.d -O2 -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/lib/libiconv-stub/include -I/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/lib/libintl-stub/include -ffile-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0=qemu-4.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -c -o util/mmap-alloc.o util/mmap-alloc.c
util/mmap-alloc.c: In function 'qemu_ram_mmap':
util/mmap-alloc.c:144:26: error: 'MAP_SYNC' undeclared (first use in this function); did you mean 'MS_SYNC'?
  144 |         map_sync_flags = MAP_SYNC | MAP_SHARED_VALIDATE;
      |                          ^~~~~~~~
      |                          MS_SYNC
util/mmap-alloc.c:144:26: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0/rules.mak:69: util/mmap-alloc.o] Error 1
make[3]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/qemu-4.2.0'
```

Note that musl undefines MAP_SYNC for MIPS and PowerPC.

closed 2020-03-30 by yousong