openwrt/packages #11083: ddns-scripts ddns-scripts_cloudflare.com-v4 Cloudflare reports invalid TTL

ImoutoChan, 2020-01-21:
Maintainer: couldn't find
Environment: Xiaomi Mi Router 3G MediaTek MT7621 ver:1 eco:3 OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152)

Description:

ddns-scripts_cloudflare.com-v4 reports an error
```
 072317  WARN : CloudFlare reported an error:
 072317       : {
  "result": null,
  "success": false,
  "errors": [
    {
      "code": 9021,
      "message": "Invalid TTL. Must be between 120 and 2,147,483,647 seconds, or 1 for automatic"
    }
  ],
  "messages": []
}
```
and don't update dns records. It worked for half year just fine with same configuration. So something was changed in their api probably. Also, I don't see any settings for TTL and can't fix it

shimco, 2020-01-21:
The same issue for me, manually setting the TTL in Cloudflare didn't resolve the issue.

Version info:
Firmware Version: OpenWrt 18.06.5 r7897-9d401013fc / LuCI openwrt-18.06 branch (git-19.309.48729-bc17ef6)
Kernel Version: 4.9.198
curl:	7.60.0-4	
ddns-scripts:	2.7.8-2	
ddns-scripts_cloudflare.com-v4: 2.7.8-2

Ansuel, 2020-01-21:
Could be a problem with cloudflare on their side.. Will check about this since i also use cloudflare for ddns

darrinsmart, 2020-01-21:
Looks like a CloudFlare issue. See https://community.cloudflare.com/t/no-longer-able-to-update-dns-records-via-api/143942/15

ImoutoChan, 2020-01-23:
Seems like issue was resolved by cloudflare itself. I'm closing it now.

closed 2020-01-23 by ImoutoChan