openwrt/packages #11090: Adblock 3.8.14 : blacklist sub-domains does not work

kousyougi, 2020-01-22:
This new version of adblock can not work with sub-domains.

**for example:**
Blacklist entry (/etc/adblock/adblock.blacklist):
ads.example.com
This entry blocks the following (sub)domains:
**working:**  http://ads.example.com/foo.gif
**not working:**  http://server1.ads.example.com/foo.gif
**working:**  https://ads.example.com:8000/

please help, thanks

neheb, 2020-01-22:
ping @dibdor

dibdot, 2020-01-23:
Works for me, please make sure to reload adblock after you've added/removed entries to your blacklist.

Example (real-live) test domains:
```
www.spiegel.de
cdn.prod.www.spiegel.de
sams.spiegel.de
```
all three (sub-) domains are accessible via nslookup without adblock.
After adding 'www.spiegel.de' to your blacklist and an adblock rerun only the latter domain is accessible.

closed 2020-01-23 by dibdot

kousyougi, 2020-01-23:
Thanks for your reply. @dibdot 
I just tried different config again. found out that

sub-domains work when **option adb_dnsvariant 'nxdomain'**
sub-domains can't work when **option adb_dnsvariant 'null (IPv4/IPv6)'**

then I change my config. It works fine now.