openwrt/packages #11095: luci-app-acme not working for wildcard certs

northorsouth, 2020-01-22:
I'm trying to create a wildcard cert using the luci interface for acme.sh. The challenge fails. When I run run-acme.sh, I can see an error saying that I'm the dns-01 challenge is required but the http-01 challenge is used (the dns challenge is required for wildcard certs). I have my cloudflare email and API key in the supplied fields, so the script should be able to use the dns challenge, not sure what's going wrong.

neheb, 2020-01-22:
ping @tohojo

tohojo, 2020-01-22:
Could you please post your config (without passwords, obviously), and the debug log output?

northorsouth, 2020-01-22:
[image]

Let me know if you need more info

northorsouth, 2020-01-22:
I'll post the output of run-acme in a sec

northorsouth, 2020-01-22:
Here's the debug output of run-acme, pretty sure I removed keys and identifying info.
Error of interest: "The supported validation types are: dns-01 , but you specified: http-01"

`acme: Running pre checks for domain.tld.
acme: port80 listens: 20639/nginx.conf 
acme: Found nginx listening on port 80; trying to disable.
acme: Waiting for nginx to stop...
acme: v4 input_rule: Chain input_rule (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:80 /* ACME */
acme: v6 input_rule: Chain input_rule (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 ACCEPT     tcp      *      *       ::/0                 ::/0                 tcp dpt:80 /* ACME */
acme: Running ACME for domain.tld
acme: Using standalone mode
logger: unrecognized option: issue
BusyBox v1.28.4 () multi-call binary.

Usage: logger [OPTIONS] [MESSAGE]

Write MESSAGE (or stdin) to syslog

	-s	Log to stderr as well as the system log
	-t TAG	Log using the specified tag (defaults to user name)
	-p PRIO	Priority (numeric or facility.level pair)
Wildcard domain
Lets find script dir.
_SCRIPT_='/usr/lib/acme/acme.sh'
_script='/usr/lib/acme/acme.sh'
_script_home='/usr/lib/acme'
Using config home:/etc/acme
https://github.com/Neilpang/acme.sh
v2.8.3
Running cmd: issue
_main_domain='domain.tld'
_alt_domains='*.domain.tld'
Using config home:/etc/acme
ACME_DIRECTORY='https://acme-v02.api.letsencrypt.org/directory'
DOMAIN_PATH='/etc/acme/domain.tld'
Using ACME_DIRECTORY: https://acme-v02.api.letsencrypt.org/directory
_init api for server: https://acme-v02.api.letsencrypt.org/directory
GET
url='https://acme-v02.api.letsencrypt.org/directory'
timeout=
_CURL='curl -L --silent --dump-header /etc/acme/http.header  -g '
ret='0'
ACME_KEY_CHANGE='https://acme-v02.api.letsencrypt.org/acme/key-change'
ACME_NEW_AUTHZ
ACME_NEW_ORDER='https://acme-v02.api.letsencrypt.org/acme/new-order'
ACME_NEW_ACCOUNT='https://acme-v02.api.letsencrypt.org/acme/new-acct'
ACME_REVOKE_CERT='https://acme-v02.api.letsencrypt.org/acme/revoke-cert'
ACME_AGREEMENT='https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf'
ACME_NEW_NONCE='https://acme-v02.api.letsencrypt.org/acme/new-nonce'
ACME_VERSION='2'
_on_before_issue
_chk_main_domain='domain.tld'
_chk_alt_domains='*.domain.tld'
Le_LocalAddress
d='domain.tld'
Check for domain='domain.tld'
_currentRoot='no'
Standalone mode.
_checkport='80'
_checkaddr
Using: netstat
d='*.domain.tld'
Check for domain='*.domain.tld'
_currentRoot='no'
Standalone mode.
_checkport='80'
[... 173 lines cut ...]
      udp-recv:<port>	groups=FD,SOCKET,RANGE,IP4,IP6,UDP
      udp-recvfrom:<port>	groups=FD,SOCKET,CHILD,RANGE,IP4,IP6,UDP
      udp-sendto:<host>:<port>	groups=FD,SOCKET,IP4,IP6,UDP
      udp4-connect:<host>:<port>	groups=FD,SOCKET,IP4,UDP
      udp4-datagram:<remote-address>:<port>	groups=FD,SOCKET,RANGE,IP4,UDP
      udp4-listen:<port>	groups=FD,SOCKET,LISTEN,CHILD,RANGE,IP4,UDP
      udp4-recv:<port>	groups=FD,SOCKET,RANGE,IP4,UDP
      udp4-recvfrom:<host>:<port>	groups=FD,SOCKET,CHILD,RANGE,IP4,UDP
      udp4-sendto:<host>:<port>	groups=FD,SOCKET,IP4,UDP
      udp6-connect:<host>:<port>	groups=FD,SOCKET,IP6,UDP
      udp6-datagram:<host>:<port>	groups=FD,SOCKET,RANGE,IP6,UDP
      udp6-listen:<port>	groups=FD,SOCKET,LISTEN,CHILD,RANGE,IP6,UDP
      udp6-recv:<port>	groups=FD,SOCKET,RANGE,IP6,UDP
      udp6-recvfrom:<port>	groups=FD,SOCKET,CHILD,RANGE,IP6,UDP
      udp6-sendto:<host>:<port>	groups=FD,SOCKET,IP6,UDP
      unix-client:<filename>	groups=FD,SOCKET,NAMED,RETRY,UNIX
      unix-connect:<filename>	groups=FD,SOCKET,NAMED,RETRY,UNIX
      unix-listen:<filename>	groups=FD,SOCKET,NAMED,LISTEN,CHILD,RETRY,UNIX
      unix-recv:<filename>	groups=FD,SOCKET,NAMED,RETRY,UNIX
      unix-recvfrom:<filename>	groups=FD,SOCKET,NAMED,CHILD,RETRY,UNIX
      unix-sendto:<filename>	groups=FD,SOCKET,NAMED,RETRY,UNIX
acme: Issuing cert for domain.tld failed. Moving state to /etc/acme/domain.tld.failed-1579717954
acme: Running post checks (cleanup).
Command failed: Not found
`

tohojo, 2020-01-23:
northorsouth <notifications@github.com> writes:

> [image]
>
> Let me know if you need more info

Hmm, it looks like your config is missing the field that sets which DNS
API to use? Maybe that's hiding under the '-- Addtional Field --'
dropdown?

northorsouth, 2020-01-23:
Whoops, that's embarrassing. I assumed the "DNS API" field would just add another set of credentials fields.

Why isn't cloudflare on this list? Is it supported?
https://github.com/Neilpang/acme.sh/tree/master/dnsapi

northorsouth, 2020-01-23:
Whoops, that's embarrassing. Got that now, and added the cloudflare script. There's a new error now:
"It seems that your api file is not correct, it must have a function named: dns_cf.sh_add"

tohojo, 2020-01-23:
northorsouth <notifications@github.com> writes:

> Whoops, that's embarrassing. I assumed the "DNS API" field would just
> add another set of credentials fields.

Yeah, that is a bit obtuse, I'll admit; I added the field by default so
hopefully it should be more obvious...

> Why isn't cloudflare on this list? Is it supported?
> https://github.com/Neilpang/acme.sh/tree/master/dnsapi

Sure:
https://github.com/Neilpang/acme.sh/blob/master/dnsapi/dns_cf.sh

northorsouth, 2020-01-23:
Yeah sorry, I found the cf script and deleted that comment. See error above though.

tohojo, 2020-01-23:
northorsouth <notifications@github.com> writes:

> Whoops, that's embarrassing. Got that now, and added the cloudflare script. There's a new error now:
> "It seems that your api file is not correct, it must have a function
> named: dns_cf.sh_add"

Lose the .sh :)

northorsouth, 2020-01-23:
Sir, thank you for putting up with me

tohojo, 2020-01-23:
northorsouth <notifications@github.com> writes:

> Sir, thank you for putting up with me

Haha, no worries! I guess it would be a bit friendlier to have a
dropdown box with the supported API types, but, well, nobody has
implemented that :)

neheb, 2020-02-07:
Is this solved?

tohojo, 2020-02-07:
I believe so, yeah...

closed 2020-02-07 by tohojo

northorsouth, 2020-02-07:
yeah, solved. thanks guys