openwrt/packages #11096: libgpiod: Python bindings missing

yeyeto2788, 2020-01-22:
Maintainer: @mhei
Environment: (MIPS 24Kc, Atheros AR9330 rev 1, 18.06)

Description:
I'd like some help on compiling libgpiod python bindings. I tried to build it my self from the device and it was impossible. More info here (https://github.com/adafruit/Adafruit_Blinka/issues/221)

Is it possible to add the bindings to the packages build? or at least a bit of help on building them on device will be helpful.

```
root@OpenWrt:/overlay/workspace/libgpiod# ./autogen.sh --enable-tools=yes --pref
ix=/usr/lib/ --enable-bindings-python CFLAGS="-I/$include_path" && make && sudo 
make install && sudo ldconfig
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'autostuff'.
libtoolize: copying file 'autostuff/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
configure.ac:170: error: Unexpanded AX_ macro found. Please install GNU autoconf-archive.
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
```

jefferyto, 2020-01-24:
Instead of compiling on your device, I suggest [using the SDK](https://openwrt.org/docs/guide-developer/using_the_sdk) to compile the package on your computer.

mhei, 2020-01-24:
It seems possible to include this in the normal package build process. At least, I managed to get the bindings compiled during the process. I will run-time test it tomorrow after I found out how to use it from Python.

yeyeto2788, 2020-01-25:
Hey @jefferyto, @mhei.

Sorry for the late reply (different time zone), @mhei glad to read that is it possible to include this in the package build.

I found this quick ussage of the bindings https://github.com/brgl/libgpiod/tree/master/bindings/python/examples but if importing the module within python3 goes well it should work.

If you need me to test it, just let me know. 

@jefferyto thanks for the info, if build has gone ok for a package it is possible also to add it in the available to install in OpenWrt as regular package?

closed 2020-01-25 by mhei

yeyeto2788, 2020-01-26:
Hey @mhei, 

Thanks for adding this into the packages, by any chance do you know when will be available on packages list?

```console
root@OpenWrt:/overlay/workspace# opkg update
Downloading http://downloads.openwrt.org/releases/18.06.2/targets/ar71xx/generic/packages/Packages.gz
Updated list of available packages in /usr/lib/opkg/lists/openwrt_core
Downloading http://downloads.openwrt.org/releases/18.06.2/targets/ar71xx/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /usr/lib/opkg/lists/openwrt_base
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /usr/lib/opkg/lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /usr/lib/opkg/lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /usr/lib/opkg/lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /usr/lib/opkg/lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
root@OpenWrt:/overlay/workspace# opkg install python3-gpiod
Unknown package 'python3-gpiod'.
Collected errors:
 * opkg_install_cmd: Cannot install package python3-gpiod.
```

mhei, 2020-01-26:
I guess the next run of the buildbots will pickup this change and the package will appear for snapshots. 
In other words, this new package is not even included in the latest release 19.07 for now, and thus also not in 18.06.2 which you're on I see in your trace above.
This is new material, I will think about whether we can add it to 19.07 but I definitely don't see it for 18.06 ...

yeyeto2788, 2020-01-26:
Ohh ok, sad that won't be included in 18.06. I'll see if I can do cross compilation then or at least install 19.07.

Thank you very much for your reply.

commit 2fdb8268 (2020-01-25): libgpiod: add package for the Python binding (fixes #11096)
commit 96c244ea (2020-01-25): Merge pull request #11117 from mhei/libgpiod-python-binding

libgpiod: add package for the Python binding (fixes #11096)