openwrt/packages #11108: ath10k-firmware-qca4019-ct-htt: package is missing from 19.07 build

fantom-x, 2020-01-23:
19.07-SNAPSHOT

I am trying to put together an image for gl-b1300 (ipq40xx) using the image builder and it is failing with the error below. I looked at the [base packages for 19.07](https://downloads.openwrt.org/releases/19.07-SNAPSHOT/packages/arm_cortex-a7_neon-vfpv4/base/) and the **htt** variants are missing for all **ath10k-firrmware-qca** supported targets. I also checked ipq806x for 19.07 and the **htt** firmware is present there.
Looking at **master**: the **ipq40xx** target does include **htt** packages.

Are the **htt** firmware files excluded from **ipq40xx** in 19.07 for a reason?

```
Collected errors:
 * opkg_install_cmd: Cannot install package ath10k-firmware-qca4019-ct-htt.
Makefile:153: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:112: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:196: recipe for target 'image' failed
make: *** [image] Error 2

```

BKPepe, 2020-01-24:
Makefile for ath10k-firmware is located in the main repository of OpenWrt.
https://github.com/openwrt/openwrt/tree/master/package/firmware/ath10k-firmware

I'd suggest opening an issue in OpenWrt FlySpray:
https://bugs.openwrt.org/

Closing.

closed 2020-01-24 by BKPepe

mdPlusPlus, 2020-05-29:
Note: Not fixed as of 19.07.3.

mdPlusPlus, 2021-02-03:
Just a reminder: This still doesn't work on the 19.07.6 imagebuilder with `gl-b1300`.

    Collected errors:
     * opkg_install_cmd: Cannot install package ath10k-firmware-qca4019-ct-htt.
    make[2]: *** [Makefile:158: package_install] Error 255
    make[1]: *** [Makefile:118: _call_image] Error 2
    make: *** [Makefile:208: image] Error 2