openwrt/packages #11111: ttyd: missing dependencies for libwebsockets.so.14

Legogris, 2020-01-23:
Maintainer: @tsl0922 
Environment: x86_64_musl

Description:　`make` fails:
```
make[3]: Entering directory '/home/legogris/dev/openwrt/feeds/packages/utils/ttyd'
mkdir -p /home/legogris/dev/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/legogris/dev/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/.pkgdir/ttyd/. /home/legogris/dev/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/legogris/dev/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.ttyd_installed
mkdir -p /home/legogris/dev/openwrt/bin/targets/x86/64/packages /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd/CONTROL /home/legogris/dev/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd/usr/bin
install -m0755 /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-install/usr/bin/ttyd /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd/usr/bin/
install -d -m0755 /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd/etc/config/
install -m0600 ./files/ttyd.config /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd/etc/config/ttyd
install -d -m0755 /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd/etc/init.d/
install -m0755 ./files/ttyd.init /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd/etc/init.d/ttyd
find /home/legogris/dev/openwrt/build_dir/target-x86_64_musl/ttyd-1.5.2/ipkg-x86_64/ttyd -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package ttyd is missing dependencies for the following libraries:
libwebsockets.so.14
make[3]: *** [Makefile:63: /home/legogris/dev/openwrt/bin/packages/x86_64/packages/ttyd_1.5.2-3_x86_64.ipk] Error 1
```
`libwebsockets-full` selected in menuconfig.

rahmadhaninovianjaya, 2020-01-24:
<html>
   <head>
     <title>Rahmadhani Novian Jaya </title>
   </head>
   <body>
   		<h3>https://bughunt-borneo.org</h3>
   	 	<h4>ClickJacking</h4>
		<p>Test bug ClickJacking & Stored XSS serta HTML Injeksi</p>
     <iframe src="https://rahmadhaninovianjayaa.com" height="1200px" width="1200px" style="border: 2px"></iframe>
   </body>
</html>

neheb, 2020-01-24:
Ugh I get a different issue:

```
  194 |             if (!len || (SSL_get_verify_result((SSL *) in) != X509_V_OK)) {
      |                          ^~~~~~~~~~~~~~~~~~~~~
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:194:49: error: 'SSL' undeclared (first use in this function)
  194 |             if (!len || (SSL_get_verify_result((SSL *) in) != X509_V_OK)) {
      |                                                 ^~~
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:194:49: note: each undeclared identifier is reported only once for each function it appears in
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:194:54: error: expected expression before ')' token
  194 |             if (!len || (SSL_get_verify_result((SSL *) in) != X509_V_OK)) {
      |                                                      ^
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:194:63: error: 'X509_V_OK' undeclared (first use in this function)
  194 |             if (!len || (SSL_get_verify_result((SSL *) in) != X509_V_OK)) {
      |                                                               ^~~~~~~~~
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:195:27: warning: implicit declaration of function 'X509_STORE_CTX_get_error' [-Wimplicit-function-declaration]
  195 |                 int err = X509_STORE_CTX_get_error((X509_STORE_CTX *) user);
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:195:53: error: 'X509_STORE_CTX' undeclared (first use in this function)
  195 |                 int err = X509_STORE_CTX_get_error((X509_STORE_CTX *) user);
      |                                                     ^~~~~~~~~~~~~~
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:195:69: error: expected expression before ')' token
  195 |                 int err = X509_STORE_CTX_get_error((X509_STORE_CTX *) user);
      |                                                                     ^
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:196:29: warning: implicit declaration of function 'X509_STORE_CTX_get_error_depth' [-Wimplicit-function-declaration]
  196 |                 int depth = X509_STORE_CTX_get_error_depth((X509_STORE_CTX *) user);
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:196:77: error: expected expression before ')' token
  196 |                 int depth = X509_STORE_CTX_get_error_depth((X509_STORE_CTX *) user);
      |                                                                             ^
/home/mangix/devstuff/openwrt/build_dir/target-mips_24kc_musl/ttyd-1.5.2/src/http.c:197:35: warning: implicit declaration of function 'X509_verify_cert_error_string' [-Wimplicit-function-declaration]
  197 |                 const char *msg = X509_verify_cert_error_string(err);
```

neheb, 2020-01-24:
This issue seems bogus. libwebsockets is a dependency. Try running make clean and then make.

Legogris, 2020-01-24:
Ah, closing it then. I ended up removing ttyd alltogether since I don't really need it.

closed 2020-01-24 by Legogris