openwrt/packages #11134: `<python3>:` round() does not truncate as expected

novski, 2020-01-26:
Maintainer: @\<commodo> 
Environment: ramips, Linkit Smart 7688, openWRT 18.6

Description:
value l/m does not get truncated to two decimals after the dot...
```
root@OpenWrt:~# python3
Python 3.7.4 (default, Sep 15 2019, 18:13:03) 
[GCC 7.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> l = round((69.2222222/100),2)                  
>>> print(l)
0.68999999999999995
>>> type(l)
<class 'float'>
>>> m = 69.22222222/100
>>> print(m)
0.69222222220000007
>>> type(m)
<class 'float'>
>>> round(m,2)
0.68999999999999995
```

[Link to stackoverflow question](https://stackoverflow.com/questions/59921794/round-does-not-trucate-float).

Quote of the comment from Mark Dickinson:

>  If the platform has some method for emulating IEEE 754-compliant arithmetic, then it should be possible to configure Python's build system appropriately to use that method. The question I linked above has a few more details.

Does this behavior affect more than my Target?

jefferyto, 2020-01-27:
I don't think this is specific to your target, though it is curious: you are using OpenWrt 18.06, the current available version of Python 3 for 18.06 is [3.6.10](https://github.com/openwrt/packages/blob/openwrt-18.06/lang/python/python3-version.mk#L9-L11) but your example output says Python 3.7.4.

It turns out that the Python configure script wasn't detecting float endianness correctly when cross-compiling. This was fixed by [bpo-34585](https://bugs.python.org/issue34585) but this was only applied to 3.8, meaning Python 2 in all versions and Python 3 in 18.06 and 19.07 are still affected.

I'm not too concerned about Python 2 in master, but for 18.06 and 19.07 I wonder if we should backport the patch for Python 3 (and Python 2 if it isn't too much work). I also wonder why upstream didn't backport the fix to their 3.6 and 3.7 branches. @commodo what do you think?

commodo, 2020-01-27:
i try a few cherry-picks;

python2 looks pretty difficult with that patch (i think this one https://github.com/python/cpython/commit/b3b8cb419e496629873fa7dda82a01863f58617a )

3.7 applied with no hassle
3.6 needs 1 minor conflict resolution;

though the noisiness of the patch could make someone skeptical to accept it in release branches;
from my side it's fine to backport;
maybe we could go for 19.07 as backporting, but 18.06..... it may be time to start to ask people to upgrade;

jefferyto, 2020-01-27:
It might be be easier to cherry pick python/cpython@b3b12c491b4fa211e8cf8743f69ad6a27421ba58 then run autoconf manually to generate a second patch, rather than cherry picking python/cpython@b3b8cb419e496629873fa7dda82a01863f58617a directly.

> though the noisiness of the patch could make someone skeptical to accept it in release branches;

Do you mean submitting the patches upstream?

> maybe we could go for 19.07 as backporting, but 18.06..... it may be time to start to ask people to upgrade;

It seems 18.06 is now in "security maintenance" mode ([wiki](https://openwrt.org/docs/guide-developer/security#support_status), [openwrt-devel thread](https://lists.infradead.org/pipermail/openwrt-devel/2020-January/021299.html)), so I would think 18.06 packages would also only receive security updates, though I'm not sure if this is the official policy :thinking:

commodo, 2020-01-28:
> Do you mean submitting the patches upstream?

no no; i meant 18.06 & 19.07

jefferyto, 2020-01-31:
This should be fixed by #11175 and #11176. (I don't think we will patch 18.06 as this isn't a security issue.)

novski, 2020-02-02:
Pleased to read this! Thanks a lot!

neheb, 2020-02-08:
SInce it's fixed, closing.

closed 2020-02-08 by neheb