openwrt/packages #11135: Problem with adduser

alisaeed, 2020-01-27:
Hi
My branch is 19.07,When I added adduser in my image OS, I could not change pasword with passwd and I have below error:

`passwd: failed to crypt password with salt '$6$WvRuLtmif./E3ps': Function not implemented
The password for test is unchanged.`

But when I did not use adduser I can change my password.

neheb, 2020-01-28:
I think the problem is https://github.com/openwrt/packages/pull/10818

It seems @BKPepe backported it.

That patch seems totally wrong. It relies on https://github.com/openwrt/openwrt/pull/1331 which is disabled by default.

RonnyReporter, 2020-01-31:
Having the same issue building current branch x86-64 images. Never happened in the past. Hope this can be fixed soon.

neheb, 2020-01-31:
I looked at the codebase briefly. chpasswd supports the -m parameter, which uses md5crypt to encrypt the password. This is supported as the size hack only applies to SHA algorithms.

The passwd utility does not.

neheb, 2020-01-31:
One way to fix this is to change /etc/login.defs 's ENCRYPT_METHOD to DES or MD5.

closed 2020-02-03 by neheb

KuoEuran, 2023-07-28:
Hi @neheb 
     Can I use sha256 or sha512?