openwrt/packages #11148: [19.07] dnsdist: can not be compiled with enabled ASLR PIE

BKPepe, 2020-01-28:
Maintainer: @James-TR 
Environment: aarch64_cortex-a53, Turris MOX, OpenWrt 19.07 (perhaps OpenWrt master as well)

Description:

When I enabled ASLR PIE for the full build, I was not able to compile dnsdist unless I add `PKG_ASLR_PIE:=0` to the Makefile.

Faillog:
```
dnsdist.o: In function `std::type_info::operator==(std::type_info const&) const [clone .isra.30]':
dnsdist.cc:(.text+0x20): relocation truncated to fit: R_AARCH64_LD64_GOTPAGE_LO15 against symbol `strcmp' defined in .text section in /mypath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/libc.so
dnsdist.cc:(.text+0x20): warning: Too many GOT entries for -fpic, please recompile with -fPIC
/mypath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
Makefile:1048: recipe for target 'dnsdist' failed
make[5]: *** [dnsdist] Error 1
```

neheb, 2020-01-28:
This can probably be mitigated with

CONFIGURE_ARGS += --with-pic

BKPepe, 2020-01-28:
Nope. That does not help.

neheb, 2020-01-28:
At this point I wonder if https://github.com/openwrt/openwrt/blob/master/rules.mk#L80 should be extended to mips64. I've seen this -fpic error before.

neheb, 2020-02-10:
Fixed and backported to 19.07.

closed 2020-02-10 by neheb