openwrt/packages #11150: adblock: "Whitelist" button not working

Legogris, 2020-01-28:
Maintainer: @dibdot 
Environment: x86_64, 19.07

Description:
When clicking the "Whitelist" button in the luci-app. the page hangs and as it turns out it's querying a non-existent endpoint:

```
Source map error: Error: request failed with status 404
Resource URL: https://x.x.x.x/cgi-bin/luci/admin/services/adblock/report
Source Map URL: ../sourcemaps/inpage.js.map
```

Legogris, 2020-01-28:
It seems like the domain is actually put in `/etc/adblock/adblock.whitelist`, but the query is still blocked.

Legogris, 2020-01-28:
It seems like this might be something else than just the luci app. Despite doing `service adblock restart`, `service adblock reload` and then a full router reboot, the domain still shows up as blocked despite bing present in `/etc/adblock/adblock.whitelist`.

Config for adblock:
```
config adblock 'global'
        option adb_basever '3.8'
        option adb_enabled '1'
        option adb_dns 'unbound'
        option adb_dnsvariant 'nxdomain'
        option adb_fetchutil 'curl'
        option adb_trigger 'wan6'

config adblock 'extra'
        option adb_debug '0'
        option adb_report '1'
        option adb_maxqueue '4'
        option adb_nice '0'
        option adb_dnsfilereset 'false'
        option adb_backupdir '/tmp'
        option adb_forcedns '1'
[...sources...]
```

Legogris, 2020-01-28:
This is strange. Not working even after suspending adblock. (Verified process not running).
I get nxdomain on my client (no caching in place there) but I get a result when doing `unbound-control lookup` on the domain in question.

Legogris, 2020-01-28:
OK, there's something up with the DNS of the domain itself (kernel-handbook.alioth.debian.org). But the initially reported 404 still stands, I guess, so leaving this open.

dibdot, 2020-01-28:
Please reset you luci caches and try again (`rm -rf /tmp/luci-*`)

dibdot, 2020-02-02:
no more feedback - closing now.

closed 2020-02-02 by dibdot