openwrt/packages #11153: [19.07] softethervpn can not be compiled with ASLR PIE

BKPepe, 2020-01-28:
Maintainer: @fededim 
Environment: aarch64_cortex-a53, Turris MOX, OpenWrt 19.07 (perhaps OpenWrt master as well)

Description:

With enabled `ASLR PIE` for the build, I am not able to compile package **softethervpn**.

Build log:
```
In file included from ./src/Mayaqua/Mayaqua.h:274:0,
                 from src/hamcorebuilder/hamcorebuilder.c:131:
/mypath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^~~~~~~
tmp/objs/Mayaqua/Cfg.o: In function `FreeCfgRw':
Cfg.c:(.text+0xa4): relocation truncated to fit: R_AARCH64_LD64_GOTPAGE_LO15 against symbol `FileClose' defined in .text section in tmp/objs/Mayaqua/FileIO.o
Cfg.c:(.text+0xa4): warning: Too many GOT entries for -fpic, please recompile with -fPIC
/mypath/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
src/makefiles/linux_64bit.mak:318: recipe for target 'tmp/hamcorebuilder' failed
make[3]: *** [tmp/hamcorebuilder] Error 1
```

neheb, 2020-02-22:
I looked at this package before. I think ASLR should be completely disabled.

poranje, 2020-02-23:
Think ?
Please share the arguments.

neheb, 2020-02-23:
PIE is not enabled by default. Probably for a good reason.

BKPepe, 2020-02-23:
> PIE is not enabled by default. Probably for a good reason.

That's why OpenWrt devs are working to have it enabled for more and more packages, right? On openwrt-devel there was suggested to a similar issue to try to add `PKG_ASLR_PIE_REGULAR:=1` to the Makefile. I didn't get to check it out yet if it works. Anyway enabled ASLR can sometimes reveal issues with some flags  (like C/LDFLAGS) that are not passed correctly. Yes, it is not enabled by default for now and it might be enabled in the future for some packages or for all. We will see.

But for now, ASLR_PIE can be disabled for this package as workaround.

closed 2020-03-11 by neheb