openwrt/packages #11163: Circle CI passed even it shouldn't

BKPepe, 2020-01-28:
Maintainer: @champtar @thess

In PR #11162 , there is wrong commit subject and each commit must start with `package name: subject` according to [CONTRIBUTING.md](https://github.com/openwrt/packages/blame/master/CONTRIBUTING.md#L42), but `buildpr` check passed.

You can take a look log from [Circle CI](https://circleci.com/gh/openwrt/packages/7012?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-checks-link):
```
=== Checking commit 'c92bb340abf0a7fbfb4529c8ad83cc3827b5b927'
Author name (Darryl Sokoloski) seems ok
Commit subject line seems ok (Maintainer: Darryl Sokoloski / @dsokoloski Compile tested: arm_cortex-a15_neon-vfpv4, TP-Link Archer C2600, master Run tested: TP-Link Archer C2600)
Signed-off-by match author
```

Looking at https://github.com/openwrt/packages/blob/master/.circleci/config.yml#L46, I think there is wrong.

champtar, 2020-01-28:
The check is pretty simple, we don't try to find the package name, not sure if there is a reliable way to do that

hnyman, 2023-01-14:
Old CI item, probably stale by now.

closed 2023-01-14 by hnyman