openwrt/packages #11172: prometheus-node-exporter-lua: option listen_interface

mistepien, 2020-01-29:
/etc/config/prometheus-node-exporter-lua

> config prometheus-node-exporter-lua 'main'
>	option listen_ipv6 '0'
>	option listen_port '9100'
>	option listen_interface 'lo'

log:
>daemon.info prometheus-node-exporter-lua: defering start until listen interface lo becomes ready

'lo' can be changed to eth0.1 and so on but it changes nothing.

champtar, 2020-01-29:
The interface needs to be an OpenWrt interface name, can you share /etc/config/network ?

mistepien, 2020-01-29:
thanks a lot, you are right.
thus it can 'lan' or 'loopback'.

closed 2020-01-29 by mistepien

jakubgs, 2021-09-19:
I'm trying to use a ZeroTier interface, and I have it in my `/etc/config/network`:
```
 # grep -A2 zerotier /etc/config/network           
config interface 'zerotier'
	option proto 'none'
	option device 'ztxyzabcxyz'
```
But for some reason the service won't start:
```
 # logread -e prometheus-node-exporter-lua | tail -n3
Sun Sep 19 23:20:34 2021 daemon.info prometheus-node-exporter-lua: defering start until listen interface zerotier becomes ready
Sun Sep 19 23:21:21 2021 daemon.info prometheus-node-exporter-lua: defering start until listen interface zerotier becomes ready
Sun Sep 19 23:21:56 2021 daemon.info prometheus-node-exporter-lua: defering start until listen interface zerotier becomes ready
```
I've created that interface as `unmanaged` since the Zerotier VPN service manages it. Is that the issue?

jakubgs, 2021-10-02:
After looking at the `/etc/init.d/prometheus-node-exporter-lua` script I found this section:
```sh
	config_load prometheus-node-exporter-lua.main
	config_get interface "main" listen_interface "loopback"
	config_get_bool ipv6 "main" listen_ipv6 0
	config_get port "main" listen_port 9100

	if [ "$interface" = "*" ]; then
		[ "$ipv6" = 1 ] && bind="::" || bind="0.0.0.0"
	else
```
Which allowed me to realize that `*` is a valid value for `listen_interface` setting, which works fine:
```
root@xyz:~ % sudo netstat -lpnt | grep 9100
tcp        0      0 0.0.0.0:9100            0.0.0.0:*               LISTEN      26798/lua
```