openwrt/packages #11174: ocserv - can not be compiled with enable seccomp

prowong, 2020-01-30:
After this commit: https://github.com/openwrt/packages/commit/072fcb5ce4938a962e2ec88d3e69dbd2b9f66e73 , ocserv can't be compiled with seccomp enabled .
  
`In file included from worker-privs.c:30:0:
worker-privs.c: In function 'disable_system_calls':
worker-privs.c:47:46: error: '__SNR_ppoll' undeclared (first use in this function); did you mean '__NR_ppoll'?
  ret = seccomp_rule_add(ctx, SCMP_ACT_ALLOW, SCMP_SYS(name), __VA_ARGS__); \
                                              ^
worker-privs.c:99:2: note: in expansion of macro 'ADD_SYSCALL'
  ADD_SYSCALL(ppoll, 0);
  ^~~~~~~~~~~
worker-privs.c:47:46: note: each undeclared identifier is reported only once for each function it appears in
  ret = seccomp_rule_add(ctx, SCMP_ACT_ALLOW, SCMP_SYS(name), __VA_ARGS__); \
                                              ^
worker-privs.c:99:2: note: in expansion of macro 'ADD_SYSCALL'
  ADD_SYSCALL(ppoll, 0);
  ^~~~~~~~~~~
worker-privs.c: At top level:
cc1: warning: unrecognized command line option '-Wno-stringop-truncation'
Makefile:1712: recipe for target 'worker-privs.o' failed
`
Thank you.

prowong, 2020-02-02:
Delete this line `$(CP) $(PKG_INSTALL_DIR)/usr/include/seccomp-syscalls.h $(1)/usr/include/` in the Makefile of libseccomp can work. Maybe it should be made a switch in the file .

neheb, 2020-02-08:
Sounds bogus.

prowong, 2020-02-08:
Sorry for my poor english, but it's true.  The libseccomp is dependened by ocserv with seccomp enabled.

neheb, 2020-02-08:
The solution is not correct. I don't know much about the package.

prowong, 2020-02-21:
I found [this](https://github.com/openconnect/ocserv/commit/935818346d10f47fcea276b1243a6515257ef513). It works.  Update ocserv to the latest version may solve this problem.

nmav, 2020-02-26:
The problem is a regression in libseccomp which was worked-arounded by ocserv. Thank you @prowong for identifying the issue.

closed 2020-02-26 by nmav

prowong, 2020-02-29:
Thanks for your great work, thank you very much.

commit c3ff03a3 (2020-02-26): ocserv: updated to 0.12.6

Fixes: #11174