openwrt/packages #11177: miniupnpd.hotplug - last line seems to be wrong

FunnyAnimal, 2020-01-30:
See https://github.com/openwrt/packages/blob/9aeb5cfbd402602f747477b3818edf73d11772e7/net/miniupnpd/files/miniupnpd.hotplug#L33

Shouldn't that be:

  `grep -q "ext_ifname=$ifname" "$tmpconf" && /etc/init.d/miniupnpd restart`

Just tested: when I change '||' to '&&', the UPNP daemon correctly restarts after doing `/etc/init.d/network restart`, otherwise it will not, flooding logread with lines as described in https://forum.openwrt.org/t/miniupnp-error-message-flooding-syslog/24446

jow-, 2020-01-30:
No, the logic of the last line is correct. The miniupnpd service hsall only be restarted if the determined wan interface is *not* already in the generated config. Your inverted logic just works around another root cause.

FunnyAnimal, 2020-01-30:
Ok, thanks jow (also for the quick response :) )

But the issue is that on my system, miniupnpd never restarts when the network is restarted. As a consequence, logread is flooded with messages ...

neheb, 2020-02-12:
This should fix it: https://github.com/openwrt/packages/pull/11275

neheb, 2020-02-13:
Duplicate of https://github.com/openwrt/packages/issues/8303

closed 2020-02-13 by neheb